subsurface/subsurface #2535: Hang during startup with cloud storage sync and update consent modals

tdyas, 2020-01-14:
### Describe the issue:
- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
The cloud storage modal and the update check consent modal both appeared during Subsurface startup and "hung" my KDE system. I suspect they captured all input. I could not interact with buttons on either modal though to cancel or respond to either of them. I had to login via a virtual terminal to kill Subsurface to free up my KDE session. Screenshot attached.

### Operating system:
- Ubuntu 19.10
- KDE Plasma 5.17.5 / KDE Frameworks 5.65.0 / Qt 5.12.4

### Subsurface version:
- Subsurface 4.9.3

### Steps to reproduce:
Unclear how to reproduce. I've attached a screenshot demonstrating what happened.

### Current behavior:
The system hung.

### Expected behavior:
Don't hang if both modals appear concurrently.

[image]

dirkhh, 2020-01-30:
I have a hard time reproducing this, but I remember that I've seen something similar once myself (but then it was two dialogs that both required input). Here I would have assumed that worst case once the cloud sync completes, everything goes back to normal. Is that not the case?

tdyas, 2020-03-28:
It wasn't in this case. On two subsequent occasions, both dialogs appeared again but in both cases I was able to click the Accept button on the update consent dialog and there was no hang.

mikeller, 2026-10-17:
Hi @tdyas,

This matches a recurring pattern we've seen with concurrent startup dialogs (consent/update modals) interfering with cloud sync at launch, also reported in a couple of later issues, but it's a timing-dependent race we've never been able to reliably reproduce or pin down. Given this report is now 6 years old with no further detail since, we're closing it as stale rather than leaving it open indefinitely.

Please reopen if you're still seeing a startup hang with concurrent modals on a current build, a fresh, detailed repro (ideally with a verbose log) would help far more than this old report can on its own.

closed 2026-10-17 by mikeller