subsurface/subsurface #2541: Proposal: simplify C++/QML interface and more obvious to use.

janiversen, 2020-01-17:
I think it is correct to say that most if not all of the active committers/contributors find the C++/QML interface no less than suboptimal (much more harsh words have been used).

I too find our C++/QML interface less than suboptimal, but I see ways to improve it, even though it will never be as elegant or robust as a direct C++ interface.

Late autumn 2019 I was very frustrated nailing down QML bugs while keeping our version max. 2-3 weeks behind subsurface master, which often ended up in replace PrefX with PrefY or changing the content of a string “Imperial” -> “imperial”, all due to the fact that QML both at compile time and runtime only makes very rudimentary validity check on C++ objects. I came up with a rather radical idea, which I earlier presented the enum part. While the radical solution was enough, when not considering the whole project, it is by far not the best way for an integrated solution.

Having had the possibility to pick the brain of a friend of mine, who is a successful consultant on large scale Qt/QML projects, I made a design, and present this as a proposal, to bring us forward.

A couple of numbers to highlight the current C++/QML interface situation:
- Q_PROPERTY (only used to create QML variables) is present in 31 header files
- Q_INVOKABLE (only used to make C++ functions available to QML) is present in 31 header files
- there are 37 objects registered to be available in QML.
- Q_OBJECT is present in 165 header files. Remark Q_OBJECT is needed for signal/slot handling, but without the QML interface it could be reduced to around 70-80 classes.

This present (especially the 37 objects) a lot of error possibilities in QML, and a lot of frustration when writing QML code.

The objective of this proposal is (in priority order)
- Simplify the C++/QML interface, make it less error prune at easier to use
- Make the least possible changes in code outside mobile-widgets
- Implement on a “is actually needed” basis (in contrast to qPref that uses “make all available” basis)
Many of my contributions have, apart from making enhancements to the mobile version, also added a lot to the complexity, hence it is natural I offer to do the implementation if/when a modified version of this proposal is accepted.

— Proposal —
Make 1 or 2 pure interface classes located in mobile-widgets.
The interface classes contain
- all Q_PROPERTY for variables actually used in QML
- All Q_INVOKABLE for all functions actually used in QML
- The code is restricted to either a function call to a backend function or a reference to a global variable like prefs. (This makes the class a pure interface function)
- Enums used in QML are duplicated in the interface class (due to lacking functionality in the Q_ENUM macro)
This is implemented in parallel with the existing code, in order to activate it following procedure is suggested:
- Search/replace QML for each C++ object, 1 commit pr object (to make testing easier)
- Delete (and only delete) Q_PROPERTY and Q_INVOCABLE in the rest of the system (only deleting will still touch about 31 files, but are easy to review, and without side effect on the remainder of the system).

Advantages:
- There are only 1 or 2 objects exposed to QML, which are only changed when adding new functionality to QML.
- Using Q_ENUM gives compiler warnings, instead as with strings being valid
- Code outside mobile-widgets (the 31 classes) becomes simpler.
- Easy to maintain and expand when needed.
- Separation between functional code and interface code, making bisecting easier (today a commit can affect both the interface and the functional behavior)
- A slim easy to read C++/QML interface

Disadvantages:
- No immediate effect on the user experience
- Touching a larger number of files

I look forward to hear comments and discuss modifications. In case the idea is accepted a series of PRs can be presented very quickly.

Thanks for taking the time to read my proposal

dirkhh, 2020-01-17:
Jan, thank you for the time you took to lay out and explain both the rationale behind your idea and the fundamental design that you propose. I appreciate the amount of effort you are putting into making Subsurface-mobile better.

When Subsurface-mobile was started, there was a single class the QML code interacted with, the `QMLManager`. A LOT of different things were all done through this ONE class that was exposed to QML. The class grew huge and started accessing things all over the place. Other contributors started working on Subsurface-mobile and found this class to be a complete mess and started to add other classes that talked to QML.

One of the more recent examples for this: we used to have a very odd but frankly straight forward way to access the `prefs` data structure; the problem was that this way didn't correctly propagate change to the QML code. Instead of adding a simple signaling mechanism that would notify the QML UI of such changes, the decision was made to add a plethora of `qPref*` classes. Actually, of the 31 classes that define `Q_PROPERTY` more than half (16) are the `qPref*` classes that were added. And IIRC it was all the churn around that code that caused a lot of the frustration last year and that in the end turned away a couple of long time Subsurface developers who never came back.

So my honest and humble request here is that we don't repeat that experience. I hate the way `qPref*` turned out, I hate the noise that it adds everywhere in our code, I hate the macros that are used to construct things and that completely obfuscate the way data is stored. But I absolutely and categorically refuse another round of "let's change 118 files with a ton of 'simple substitutions' just so things are prettier" changes. No, let me take that back. Not "things are prettier". Even if the new code is objectively better and cleaner and faster and more shiny and robust and whatever.

And of course those 118 files are just the incidences of `qPref`. What you propose here is far more invasive than that, as far as I can tell.

I am willing to look at small incremental changes that have an immediate user in QML and that strike a conservative balance between "this makes QML simpler" vs "this causes a lot of churn". I'm not going to invent random metrics here, but anything that causes significantly more change to existing code than "positive" change to QML code is something that I'll be concerned about. Again, this isn't a metric, I'll continue to look at any PR that you send. But I really don't want to go down another round of changes to hundreds of existing files just because it makes things cleaner for QML.

janiversen, 2020-01-17:
I think I did not explain an important part very well. The interface class(es) can be implemented without touching a single file outside mobile-widgets ! It is totally optional, but recommended, to delete Q_PROPERTY from the header files outside mobile-widgets. 

(I do not think you have forgotten, but your text indicates may be, the first very big chunk of qPref classes was contributions from me, so I painfully remember a lot of the fuzz, which eventually lead to me being told not to contribute anymore).

I find your suggestion about small steps an excellent way, since it gives an easy way to strike a balance.

I consider QMLmanager a big monolith that are quite hard to understand. To me the absence of a clear separation between implementation and interface makes it even harder. The nice thing about a pure interface class(es) is that it contains no implementation but merely function calls to the "real" code.

dirkhh, 2020-01-17:
I very much remembered that, but I intentionally didn't want my response to read like an accusation, so I rewrote this a couple of times to not make it about who made those changes.

janiversen, 2020-01-17:
A good point. I to was not impressed by the way qPref turned out, but like you, I do not feel it is worth the pain to do another implementation.

janiversen, 2020-01-22:
Solved.

closed 2020-01-22 by janiversen