subsurface/subsurface #2552: SegVault when adding gas change to unsaved cylinder

djunkins, 2020-01-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

If you add a new cylinder in the Equipment tab and <control>-click on the dive profile to add a gas change before clicking "Apply Changes" in the equipment tab, the new cylinder appears in the list of cylinders that the gas can be changed to, but when you select the new cylinder the program crashes with a Segmentation Fault. The program does not crash if you hit "Apply Changes" and then switch the gas.

### Operating system:

MacOS 10.14.6 (English)

### Subsurface version:

4.9.3-798-g1f51251f1b80
Master branch hash 1f51251f1

### Steps to reproduce:

1) Select a dive and go to Equipment tab
2) Add a new cylinder but don't click "Apply Changes"
3) Move to the dive profile and <control>-<click> to bring up the menu
4) Select <Add Gas Change> and click on the new cylinder that appears in the list of available gases
5) Program will crash with a segmentation fault.
### Current behavior:

### Expected behavior:

I would not expect the new cylinder to appear the list of available gases until after the "Apply Changes" button is clicked.
[subsurface-crash-report.txt](https://github.com/Subsurface-divelog/subsurface/files/4105695/subsurface-crash-report.txt)

### Additional information:

### Mentions:

dirkhh, 2020-01-30:
Hey @bstoeger  -- this looks like something that might be in your comfort zone :-)

bstoeger, 2020-01-30:
> this looks like something that might be in your comfort zone

I don't think so.

In any case, I can't compile current master:

```
[  9%] Building CXX object core/CMakeFiles/subsurface_corelib.dir/qtserialbluetooth.cpp.o
/home/bs/src/subsurface/core/qtserialbluetooth.cpp: In function ‘dc_status_t ble_packet_open(dc_iostream_t**, dc_context_t*, const char*, void*)’:
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:301:3: error: invalid conversion from ‘dc_status_t (*)(void*, int)’ to ‘dc_status_t (*)(void*, unsigned int, unsigned int, dc_parity_t, dc_stopbits_t, dc_flowcontrol_t)’ [-fpermissive]
  301 |   qt_ble_poll, /* poll */
      |   ^~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, int)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:304:3: error: invalid conversion from ‘dc_status_t (*)(void*, unsigned int, void*, size_t)’ {aka ‘dc_status_t (*)(void*, unsigned int, void*, long unsigned int)’} to ‘dc_status_t (*)(void*)’ [-fpermissive]
  304 |   qt_ble_ioctl, /* ioctl */
      |   ^~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, unsigned int, void*, size_t) {aka dc_status_t (*)(void*, unsigned int, void*, long unsigned int)}
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:307:3: error: invalid conversion from ‘dc_status_t (*)(void*, unsigned int)’ to ‘dc_status_t (*)(void*)’ [-fpermissive]
  307 |   qt_custom_sleep, /* sleep */
      |   ^~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, unsigned int)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:308:3: error: invalid conversion from ‘dc_status_t (*)(void*)’ to ‘const char* (*)(void*)’ [-fpermissive]
  308 |   qt_ble_close, /* close */
      |   ^~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp: In function ‘dc_status_t rfcomm_stream_open(dc_iostream_t**, dc_context_t*, const char*)’:
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:333:3: error: invalid conversion from ‘dc_status_t (*)(void*, size_t*)’ {aka ‘dc_status_t (*)(void*, long unsigned int*)’} to ‘dc_status_t (*)(void*, unsigned int*)’ [-fpermissive]
  333 |   qt_serial_get_available, /* get_received */
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, size_t*) {aka dc_status_t (*)(void*, long unsigned int*)}
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:335:3: error: invalid conversion from ‘dc_status_t (*)(void*, int)’ to ‘dc_status_t (*)(void*, unsigned int, unsigned int, dc_parity_t, dc_stopbits_t, dc_flowcontrol_t)’ [-fpermissive]
  335 |   qt_serial_poll, /* poll */
      |   ^~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, int)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:338:3: error: invalid conversion from ‘dc_status_t (*)(void*, unsigned int, void*, size_t)’ {aka ‘dc_status_t (*)(void*, unsigned int, void*, long unsigned int)’} to ‘dc_status_t (*)(void*)’ [-fpermissive]
  338 |   qt_serial_ioctl, /* ioctl */
      |   ^~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, unsigned int, void*, size_t) {aka dc_status_t (*)(void*, unsigned int, void*, long unsigned int)}
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:340:3: error: invalid conversion from ‘dc_status_t (*)(void*, dc_direction_t)’ to ‘dc_status_t (*)(void*, unsigned int)’ [-fpermissive]
  340 |   qt_serial_purge, /* purge */
      |   ^~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, dc_direction_t)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:341:3: error: invalid conversion from ‘dc_status_t (*)(void*, unsigned int)’ to ‘dc_status_t (*)(void*)’ [-fpermissive]
  341 |   qt_custom_sleep, /* sleep */
      |   ^~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*, unsigned int)
/home/bs/src/subsurface/core/qtserialbluetooth.cpp:342:3: error: invalid conversion from ‘dc_status_t (*)(void*)’ to ‘const char* (*)(void*)’ [-fpermissive]
  342 |   qt_serial_close, /* close */
      |   ^~~~~~~~~~~~~~~
      |   |
      |   dc_status_t (*)(void*)
make[2]: *** [core/CMakeFiles/subsurface_corelib.dir/build.make:1363: core/CMakeFiles/subsurface_corelib.dir/qtserialbluetooth.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:485: core/CMakeFiles/subsurface_corelib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
I did a `git submodule update`.

When I run `build.sh` I get:
```
 ./subsurface/scripts/build.sh -build-with-webkit -desktop 
building Subsurface in subsurface/build
[... 151 lines cut ...]
config.status: creating include/libdivecomputer/Makefile
config.status: creating include/libdivecomputer/version.h
config.status: creating src/Makefile
config.status: creating src/libdivecomputer.rc
config.status: creating doc/Makefile
config.status: creating doc/doxygen.cfg
config.status: creating doc/man/Makefile
config.status: creating examples/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
CDPATH="${ZSH_VERSION+.}:" && cd /home/bs/src/subsurface/libdivecomputer && /bin/bash /home/bs/src/subsurface/libdivecomputer/missing aclocal-1.15 -I m4
/home/bs/src/subsurface/libdivecomputer/missing: line 81: aclocal-1.15: command not found
WARNING: 'aclocal-1.15' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:423: /home/bs/src/subsurface/libdivecomputer/aclocal.m4] Error 127
```

dirkhh, 2020-01-30:
recompile and install libdivecomputer...

bstoeger, 2020-01-30:
Apart from getting confused with current_dive and displayed_dive, the code seems to be not working anyway:

The cylinder-id is parsed from the text:
```
        QString gas = action->text();
        gas.remove(QRegExp(" \\(.*\\)"));
```
That means one cannot have two cylinders with the same gas mix? Am I missing something?

bstoeger, 2020-01-30:
Oh, I see that later it accesses `action->text()` again.

commit 997e7fe1 (2020-01-30): Profile: fill gas change context menu from current dive

The gas change context menu was filled from the CylindersModel.
This means that even before saving new cylinders, the user could
add a gas change event to these cylinders.

Instead, fill from the current dive.

Fixes #2552.

Reported-by: Doug Junkins <junkins@foghead.com>
closed 2020-01-30 by dirkhh