subsurface/subsurface #2554: is anybody using the Paralenz camera?

janiversen, 2020-01-24:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:

The Paralenz camera (https://www.paralenz.com/) is (of course) danish, and I would like to see if there are a general interest of getting subsurface to support it?

Please let me know if you have a Paralenz and whether or not you find it interesting to combine the paralenz sensors into a dive captured primarily by your dive computer.

I am currently evaluating reading out the paralenz via bluetooth including the video, and (somehow) combine it with the information from a dive computer.

Thanks in advance for providing opinions. 

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2020-01-25:
What kind of data besides video does it record that could be imported? Obviously, I don’t have one.

janiversen, 2020-01-25:
The new one have sensors for 
 -Pressure (depth)
- Temperature
- GPS
- Conductivity
- 9 Axis Gyroscope (mostly used to stabilise)
- Accelerometer (gives an idea of how fast you move compared to the water around you)
- Time
All collected in a dive log together with video (if active).

So basically the dive log contains the same information as a simple dive computer and some added sensors.

dirkhh, 2020-02-12:
not a Subsurface issue

closed 2020-02-12 by dirkhh