subsurface/subsurface #2555: diveplan are registered as real dives.

janiversen, 2020-01-24:
### Describe the issue:

- [ ] Bug
- [x ] Change request
- [ ] New feature request
- [ x] Discussion request

### Issue long description:

Currently a diveplan is not different from a manually entered dive, this seems wrong.

Should we not mark diveplans as bring diveplans (using e.g. divemode) and let the user decide (via a switch in setup) whether or not to consider a diveplan a real dive?

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

janiversen, 2020-01-24:
@atdotde I would really like to hear your opinion on how you use the diveplans? when diving do you also use a dive computer so you (as I) end up having 2 entries in the divelist?

atdotde, 2020-01-25:
I don’t save dive plans in my regular log but I know others do (@willemferguson ?). What is wrong with treating them no different from other dives? You can always look at the dive computer name to see it was planned. We record the relevant planner settings in the notes but don’t apply them we hen replanning but rather use the current ones. That is by design.

willemferguson, 2020-01-25:
Yes they are currently treated as normal dives. I cannot think of an application where one would want to have dive plans reflected in the dive statistics and being numbered as a real dive. If the dive times of the plan and the real dive overlap the dive plan shows up as a different dive computer for the same dive. In that case, if the plan is not the first dive computer of the dive, the dive plan is not reflected in the dive statistics. If there is no overlap it shows as a different dive, reflected in the dive statistics as well as having a dive ID number. In my opinion this is a good case for being able to mark dives (i.e. dive plans in this case) as invalid (bstoeger's proposal). That way the dive plans are not incorporated in dive statistics and dive numbering.

janiversen, 2020-01-25:
Thanks for responding.  

@atdotde 
I have a handful of dive plans saved, representing a couple of dive spots we use for training. I will have a look with the debugger to see how the dive computer name signal a dive plan (as opposed to a manually entered dive).

I think the idea of not reusing the notes are a very good idea, but thanks for telling me, I have to change that in the new editor :-(

@willemferguson
I totally agree it does not make sense to count the dive plans as dives (we have one diver who tells me differently, but we recently sponsored him with a simple dive computer). But currently they are counted. I made 4 dive plans just to test, and the yearly statistic grew by 4.

Sounds like a way to check if it is a plan or a real dive, marking the dive plans as invalid is a more secure way. This sounds simpler as my idea to extend dive_mode with an extra type "plan".

I came across the problem with doing the calculations for dive summary, let me try to apply the ideas you have given me. Thanks again.

janiversen, 2020-01-25:
@dirkhh do you have an opinion, which way is the preferred way to secure dive plans are not counted as real dives
- invalid flag
- extent dive_mode
(this has a small added advantage, that we can jump directly to the dive plan editor when opening the dive).

dirkhh, 2020-01-26:
I don't think that's really an invalid dive. It seems like it would be much better described by a different dive_mode

willemferguson, 2020-01-26:
The question is; How would a diveplan-mode integrate with the rest of the dive log? Each dive plan has an appropriate dive mode, e.g. a CCR dive plan has a dive mode of CCR in order to do the appropriate calculations and display of profile. If this is now changed to diveplan-mode, how would the dive planner know how to do calculations and profile presentation? Related to the above point, all the dive plans I have kept are those that correspond to real dives and are found as a second dive computer for a specific dive. Now we would have two dive computers with dives in different dive modes, (e.g one a real dive in CCR mode and another in dive-plan mode) which may not make a lot of sense.

atdotde, 2020-01-26:
There is already a test for planned dives in core/dive.c: 

bool is_dc_planner(const struct divecomputer *dc)

I can see that most of the time you don't want plans in your statistics but maybe people use those as entries for dives without a valid computer log.

I must say, the whole "invalid dive" thing is something I do not see an urgent need for: at least on desktop, you can have more than one log file and this allows you to have one to include your plans (as I said, I personally don't save them0 while you have another one where you keep your actual dives. Admittedly, this might be more complicated on mobile with its lack of actual files.

janiversen, 2020-01-26:
is_dc_planner() is perfect for what I need, thanks a lot, this is a lot "nicer" than extending dive_mode or the invalid flag (A feature I too do not see as urgent for the mobile).

To me statistic is one part of the problem, but I also want to have a list of diveplans in the dive planner pages, allowing the user to easy view/copy/delete a template plan, with is_dc_planner() I 
can easily do that.

I see was not precise enough, I wrote add a dive_plan type to dive_mode, what I actually meant was
add PLAN_OC, PLAN_CCR and PLAN_PSCR, exactly in order to allow the calculations to work. I believed that to be the only way.

I have been confused by how the collapsing works and now I know why I was confused. The program collapses the dive plan and the real dive, but it is done by giving using the same dive number for both, and it still shows as 2 entries in the divelog. This however is not a big problem because (for the mobile) it is easy to add is_dc_planner() to the filter.

Thanks a lot to all for helping me understand the code a bit better and help the mobile diveplanner go one step further.

closed 2020-01-26 by janiversen

janiversen, 2020-01-26:
just a short update. PR #2553 makes good use of the function.