subsurface/subsurface #2567: mobile UI: add option to limit to one column in portrait mode

dirkhh, 2020-01-30:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

This has come to me through an in-app support request.
A user would like to be able to have only one column shown when holding their iPad in portrait mode. Seems like a reasonable option to add...

### Operating system:

request came from iOS, but realistically the same is of course valid independent of OS

### Additional information:

Interestingly, the user specifically stated that they liked multiple columns in landscape format, it's just in portrait format they want to lock things to one column

### Mentions:

@janiversen - I looked at how to implement this and realized that I would be editing things in the middle of the code that you are currently working on, just making things harder for you.
So I'm posting this here. Maybe you can just throw it in with your other changes, or we can leave it pending until your current batch of changes are all in master and I can work on it then.

janiversen, 2020-01-30:
That is a nice feature, we did something very similar but only for small devices.

I think the right way to implement this is to have user selectable columns in the divelist, with an extra marker setting the default column.

I have been holding back on these type of changes because the include bigger changes to the divelist model and filter.

I think the right way to attack this is:
1) get the new divelist model/filter (withouts command) integrated in mobile
2) update the divelist qml with selectable columns.
3) add the new navigation.

agree?

I will add this to my work list (can you assign it to me)

dirkhh, 2020-01-30:
Berthold and I are working on integrating the models which will give us full undo support on mobile. This is a rather bigger set of changes and I wouldn’t want to do too much on the dive list until then (because as a result the divelist QML code gets completely rewritten as well).
The other major benefit besides real undo support is that this also massively speeds up the divelist performance for people with many dives.
This also affects the filter code (but for now we have not tried to expose the full filter semantics on mobile). All this is of course WIP.

I’m happy to wait with this feature for later - especially if we end up moving away from the PageStack.

janiversen, 2020-01-30:
I am aware of that, which is why I try to touch qt-models as little as possible, and I look very much forward to seeing the result.

If wanted when the time comes, I offer to help rewrite the QML code (a bit egoistically because it will help me integrate some features at the same time).

Having "undo" is something not seen often in mobile apps, and therefore something that should be announced a bit bigger than normal.

in order for the new navigation to be integrated some cleanup are needed in main.qml and the pages need a smaller conversion. We could target the new navigation, new divelist and a simple dive planner for a jump in the major version number (we kiddingly call our version subsurface-mobileIII).

closed 2020-02-12 by dirkhh