subsurface/subsurface #2579: Subsurface Mobile (Android) - notes field does not autoscroll while editing

mwerle, 2020-02-03:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When adding more lines while editing the note for a dive in Subsurface Mobile for Android, the field resizes to become larger but the screen does not scroll to accommodate the increased size.  The effect is that the input cursor disappears off the bottom of the screen making it impossible to see what one is typing.

### Operating system:

Samsung Galaxy Note 4 - Android Lollipop (Samsung TouchWiz)
Huawei P30 - Android Pie (Huawei EMUI)
Huawei Mediapad M5 - Android PIe (Huawei EMUI)

### Subsurface version:

Subsurface 2.2.3 (4.9.3.273)

Using official latest release for Android (beta channel). Same issue occurred with older versions of Subsurface (unsure which exact versions).

### Steps to reproduce:

1. Sync dives (either directly from dive computer or from cloud).
2. Select and edit a dive.
3. Select the "Notes" field
4. Start entering data.  As you enter more lines of data, the cursor disappears off the bottom of the screen (also happens if using an external keyboard), making further data entry difficult.

### Current behavior:

As more lines are added in the "Notes" field, the cursor moves beyond the bottom of the display instead of the display scrolling up.

### Expected behavior:

The display should scroll up if the entry point moves to the bottom of the screen. The cursor should always stay visible.

### Additional information:

### Mentions:

dirkhh, 2020-02-03:
No argument, certainly annoying.
Sadly, I have no idea how to do that with the underlying toolkit that we use (i.e., I've noticed this as well, but don't know how to fix it)

RickMWalsh, 2020-02-07:
Sorry I don't have a build environment (or time) at the moment, but could some help be on https://doc.qt.io/qt-5/qml-qtquick-controls2-textarea.html

dirkhh, 2020-02-07:
> Sorry I don't have a build environment (or time) at the moment, but could some help be on https://doc.qt.io/qt-5/qml-qtquick-controls2-textarea.html

I'm quite familiar with that documentation... we are already doing some magic with scrolling __because__ there doesn't appear to be an option for this widget to do the "right" thing (which is actually rather ill defined in this context, anyway). Specifically, if you assume that the notes field gets bigger than your screen, trying to figure out where in the TextArea your cursor might be and keeping that in the visible area above the virtual keyboard has turned out to be a lot harder than what my QML skills allow me to do.
But I'll definitely take another look.

RickMWalsh, 2020-02-07:
Thanks Dirk,

On Sat, 8 Feb. 2020, 03:46 Dirk Hohndel, <notifications@github.com> wrote:

> Sorry I don't have a build environment (or time) at the moment, but could
> some help be on https://doc.qt.io/qt-5/qml-qtquick-controls2-textarea.html
>
> I'm quite familiar with that documentation... we [...]
Looks like you worked it out and I didn't mean to imply you hadn't looked
at the documentation. Your commit 102f3bf2 fixes the problem for me on my
Galaxy S8.

>

dirkhh, 2020-02-07:
Thanks for reporting back. Much appreciated

closed 2020-02-07 by dirkhh

mwerle, 2020-02-09:
Sorry didn't get time to figure out how to set up compilation environment for Android - kept getting compiler errors in system headers.

Anyway latest Android beta version works for me.  Many thanks for looking into this and fixing it!

dirkhh, 2020-02-09:
No worries - I really appreciate the report and I appreciate your input that it's fixed.
Of course the latest beta appears to have some other issues that need attention, but that's why it's a beta and not a production version :-)