subsurface/subsurface #2593: Implementation of gradient factors.

chasep255, 2020-02-06:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

I was trying to understand the contents of deco.c and more specifically how the gradient factors are applied in buhlmann deco mode.  I am kind of confused where the following line of code comes from in the tissue_tolerance_calc.

```
/* tolerated = (tissue_inertgas_saturation - buehlmann_inertgas_a) * buehlmann_inertgas_b; */

			tissue_lowest_ceiling[ci] = (ds->buehlmann_inertgas_b[ci] * ds->tissue_inertgas_saturation[ci] - gf_low * ds->buehlmann_inertgas_a[ci] * ds->buehlmann_inertgas_b[ci]) /
						     ((1.0 - ds->buehlmann_inertgas_b[ci]) * gf_low + ds->buehlmann_inertgas_b[ci]);
```
Rewritten a little this is saying that...
c = (bp - gab) / ((1 - b)g + b)
Where
c = ceiling
p = tissue pressure
g = gradient factor
a = a value
b = b value

It is my understanding that a gradient factor is defined as...
g = (p - c) / (p - m)
Where
c = the current ceiling / depth (in terms of pressure)
p = tissue compartment pressure
m = m value = (p - a) *b

To see if I could verify / derive the formula used in subsurface I tried to transform the above equation into the one used in deco.c

Express m in terms of p, a, and b.
g = (c - p) / ((p - a)b - p)

Solve for c
c = -a b g + (b - 1) g p + p 

Then I assume if I set the two equations equal to each other I should be able to cancel out both sides.
-a b g + (b - 1) g p + p = (bp - gab) / ((1 - b)g + b)

However the only realistic solution to the above equation is when g = 1.

I have also tried to transform one into another algebraically but have had no success. Therefore I must conclude that either my understanding of what a gradient factor is / math is wrong, or there is something up with the formula used in tissue_tolerance_calc.  Can anyone take a look at this and explain where the formula comes from.

henrik242, 2020-02-06:
@atdotde probably has something to say here :)

atdotde, 2020-02-06:
Your formula for the M-value is wrong. The M-value, the maximally allowed tissue pressure at a given ambient pressure d (for depth which will be the ceiling) is given by

m = d / b + a

(this is originally expressed as d = (m - a) b). Then

g = (p - c) / (m - c) = (p - c) / (c / b + a - c) = (b p - b c) / (c + a b - b c)

This is the same as

g c + a b g - b c g= b p - b c

and so you can solve this for c as

c = (b p - a b g) / (g + b - b g)

which is the formula in the code.

And yes, it took my over an hour (including implementing the other formula) to figure this out, so it was by no means obvious.

closed 2020-02-06 by atdotde

henrik242, 2020-02-06:
@atdotde Maybe something like this could be added as comments into deco.c?