subsurface/subsurface #2595: Dive indive list does not get scope when selected

willemferguson, 2020-02-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

1) Open a dive log
2) Select a trip heading. The trip notes are displayed.
3) Select a dive from this trip. The Notes tab does not change, still shows trip notes.
    The dive notes are not shown. I would expect it show the dive notes, not the trip notes.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @bstoeger

bstoeger, 2020-02-06:
Yes - that happened recently to me as well. I will look into it.

bstoeger, 2020-02-06:
PS: Weirdly, when navigating with the cursor keys, it works - that surprised me a lot.

PR #2602 (2020-02-08): Dive list: be more careful on when updating the UI after selection
Fix two issues:

1) When narrowing the selection, we didn't get setSelection()
   calls. Only, when the user released the mouse button was
   the selection updated. Therefore, hook into the mouse-release-
   event and update the UI if the selection changed.

2) We updated the ui in setSelection(). However, this was called
   on mouse-move even if the actual selection didn't change.
   Therefore, compare selection before and after processing of
   the event and only refresh the UI if there are changes.

Clearly, this can only be [... cut]
commit 2cea115d (2020-02-08): Dive list: be more careful on when updating the UI after selection

Fix two issues:

1) When narrowing the selection, we didn't get setSelection()
   calls. Only, when the user released the mouse button was
   the selection updated. Therefore, hook into the mouse-release-
   event and update the UI if the selection changed.

2) We updated the ui in setSelection(). However, this was called
   on mouse-move even if the actual selection didn't change.
   Therefore, compare selection before and after processing of
   the event and only refresh the UI if there are changes.

Clearly, this can only b [... cut]
closed 2020-02-08 by dirkhh