subsurface/subsurface #2614: Mobile: Filter tool does not work.

willemferguson, 2020-02-11:
While the dive list is shown, press the filter icon at the bottom of the screen. On my phone the filter text box near the top of the screen is shown for about 250 ms and then disappears. This makes it impossible to use the filter tool.

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @bstoeger

dirkhh, 2020-02-12:
Which version are you testing?
This works fine for me on several devices (but all with the current beta)

willemferguson, 2020-02-12:
https://www.dropbox.com/s/tb44ej8zbvjr52y/20200212_214817.mp4?dl=0

This is latest version on Galaxy J2 with Android 7.1.1

dirkhh, 2020-02-13:
The relative size of font vs screen are... impressive :-)
I am trying to reproduce the issue in order to understand how to fix it, so far with no success (even switching to a similar size screen). I think I'll need to add some debugging output to the code in question and create a new build for you.
Can you install builds from our website, or do you need them to be on Google Play?

dirkhh, 2020-02-13:
Can you try this binary
https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.3.939-arm64.apk
and after you toggle the filter on and then off again (so tap the button twice) copy the logs in the About screen and post the log here?

willemferguson, 2020-02-14:
Apologies for trouble. My phone says:
App not installed. This app isn't compatible with your phone.
????

dirkhh, 2020-02-14:
Can you post the first ten lines of a log from your phone?

willemferguson, 2020-02-14:
I do not think there is useful information in the first 10 lines. Here is the complete log. Start looking around line 30.

[slog1.txt](https://github.com/Subsurface-divelog/subsurface/files/4203009/slog1.txt)

dirkhh, 2020-02-14:
Oh yes there is.
It shows me that you have one of the few remaining 32bit devices. I'll get you a matching binary tomorrow.

willemferguson, 2020-02-14:
Thank you. Did not realise the J2 is 32-bit.

dirkhh, 2020-02-14:
I'm sorry it took me so long. I simply have way too many things on my plate right now - and as a result I'm neither as responsive nor as diligent with code reviews as I should be.
Here's a 32bit APK that should work on your phone: https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.3.939-arm.apk
As I requested above, please install this, run it, toggle the filter on and then off again (so tap the button twice), and then copy the logs in the About screen and post them here

willemferguson, 2020-02-14:
I believe this is what you requested
[dirk2.txt](https://github.com/Subsurface-divelog/subsurface/files/4206562/dirk2.txt)

dirkhh, 2020-02-14:
Hmm. And none of my debugging info is in the log. You did tap on the filter button, right?

willemferguson, 2020-02-15:
From the previous screenmovie it appears I did download the correct Android executable (update 939). I did two trials now. The response of the Android app is not consistent. I took another screen movie but the bandwidth at the moment is so pathetic I cannot upload the .mp4 to dropbox. So here are the two log files. Does any of them provide anything useful?
[dirk3.txt](https://github.com/Subsurface-divelog/subsurface/files/4207836/dirk3.txt)
[dirk4.txt](https://github.com/Subsurface-divelog/subsurface/files/4207837/dirk4.txt)

willemferguson, 2020-02-15:
[image]

Here is a screen shot of the mobile app version

willemferguson, 2020-02-15:
Here is a link to the screen movie. The first bit refers to dirk3.txt. Then you can see me pause the screen movie and start it again. The second sequence refers to dirk4.txt

https://www.dropbox.com/s/8edigov24eljzdm/20200215_082508.mp4?dl=0

dirkhh, 2020-02-15:
@willemferguson thank you so much for all the effort you went through, including the screen movie. I am completely baffled and will try a different approach to get the information that I need. I would have expected either the debug information that I added or an error if somehow that code wasn't able to run - the fact that you get the brief flash of the filter header and none of the messages in the log means that there must be a code path I missed - more experimentation needed on my part.

willemferguson, 2020-02-15:
I will be diving for the next week. I am likely to have some Internet connectivity but I have no idea what the quality of the connection is likely to be. I will check email on my gmail account once a day.

dirkhh, 2020-02-15:
Hopefully I am catching before you leave... I think I understand what happened with the previous build. Would you please try this one?

https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.3.957-arm.apk

bstoeger, 2020-03-07:
Note that the filter also does not show up by itself for me on mobile-on-desktop. I have to force-scroll to the top to get the filter.

dirkhh, 2020-03-07:
Yes - that appears to be the answer to @willemferguson 's problem. I forgot to comment here when I ran into the same issue during testing.
What I have not figured out is when this happens. Something must be triggering it because most of the time for me I do not need to do this, but once every blue moon it happens - and for Willem it apparently happens every time...

bstoeger, 2020-03-07:
I'm in Willem's boat - it happens for me every single time on mobile-on-desktop.

dirkhh, 2020-03-21:
It appears that on some devices, but not others, in order for the filter to show up, one has to slightly pull down the dive list.
No idea why.

bstoeger, 2020-10-03:
This works for me now. Close?

willemferguson, 2020-10-03:
Yes, please close.

wf

bstoeger, 2020-10-03:
Thanks - closing.

closed 2020-10-03 by bstoeger