subsurface/subsurface #2615: Mobile export tool problems

willemferguson, 2020-02-11:
See the screenshots below. The font size in the export tool is not consistent with the rest of Subsurface-mobile, in fact the tool for getting the file name is so oversized that it is not very useful. In fact I find it impossible to specify an export file name and then exporting to that file.
This is a Samsung Galaxy J2 with Android 7.1.1

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

[image]
[image]

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @janiversen

janiversen, 2020-02-11:
The radiobuttons are easy to fix, it is just using TemplateRadiobutton instead of the “raw” radiobutton (the “raw” version ended up being used in one of the many changes requested).

For the filedialog it is a bit more complicated since it does not allow font to be set. Solution (there might be others) is to set the Treeview stylesheet in the filedialog, something like:
setStyleSheet(“QTreeView {font: 12pt \”arial\”;}”); but that needs to be done from C++
alternative use a homemade filedialog (that is how we solved iCloud connection)

unfortunately I am not in a position to provide a fix.

dirkhh, 2020-02-12:
@willemferguson  can you post a screenshot of the Developer->Theme information page?

willemferguson, 2020-02-12:
Thank you for the info, Jan.

willemferguson, 2020-02-12:
Screenshot as requested.

[image]