subsurface/subsurface #2636: Unable to import data form dive computer with Flathub-Package

thommmmmy, 2020-02-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Native rpm-package on Fedora 30 won't install. Dependency error. See: https://bugzilla.redhat.com/show_bug.cgi?id=1742889

The Flatpack-Package could be a workaround. And subsurface form Flathub is really working. But there is no way to import data from a dive computer. It's like a car, where the engine starts, but if you put in gears and push the clutch, it never starts to move. At the end of the dy, it's all most useless.

### Operating system:
Fedora 30 (Linux)

### Subsurface version:
4.8.4 form Flathub

### Steps to reproduce:
Import data form a dive computer.

### Current behavior:
Error: Unable to open /dev/ttyUSB0 Suunto (Vyper Air)

### Expected behavior:
Be able to import data form a dive computer!

### Additional information:
On earlier Fedora versions data import didn't working without root-privileges. Even that was really alarming.

dirkhh, 2020-02-24:
There shouldn't even be a Subsurface package as part of Fedora.
We distribute Fedora 30 packages here:
https://software.opensuse.org/download.html?project=home:Subsurface-Divelog&package=subsurface

closed 2020-02-24 by dirkhh

thommmmmy, 2020-02-25:
@dirkhh
Wow, works very well! Thank you very much!
But why I'm unable to find this informations on https://subsurface-divelog.org/download/ or on https://subsurface-divelog.org/faq/ ??

dirkhh, 2020-02-25:
I need to update the download page to no longer list the specific version numbers for the various distros. But the link to Fedora is right there.