subsurface/subsurface #2651: Android Mobile: DC Download Status not updated in repeated downloads

charno, 2020-03-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When repeatedly downloading from a dive computer on Android, the textual status on the right hand side of the buttons is only updated the first time.

[image]

### Operating system:
Galaxy S10 with supplied Android 10. 

### Subsurface version:
2.3.0 (4.9.3.1032) from Google Play Beta

### Steps to reproduce:
- Connect dive computer
- Press Download button
- Press Abort button
- Press Download button

### Current behavior:
On second download, the status is not updated anymore.

### Expected behavior:
On second download, the status should be updated.

### Additional information:

### Mentions:

dirkhh, 2020-03-07:
That animated GIF is awesome. Shows me exactly what you mean.
I'm curious - how did you create that?

I'll take a look and see if I can fix this.

charno, 2020-03-08:
I used https://play.google.com/store/apps/details?id=com.kimcy929.screenrecorder to record the screen and later transcoded the mp4 with `ffmpeg -i <input>.mp4 -r 15 <output>.gif`

Nice small fix! :)

commit 08e84784 (2020-03-07): mobile UI: re-enable download status when restarting download

When tapping on 'retry' we didn't clear the flag that decided which message to
display.

Fixes: #2651
commit b92b5998 (2020-03-08): mobile UI: re-enable download status when restarting download

When tapping on 'retry' we didn't clear the flag that decided which message to
display.

Fixes: #2651
closed 2020-03-08 by dirkhh