subsurface/subsurface #2657: Android-Mobile USB-Serial Devices: Allow precise device selection.

charno, 2020-03-08:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
This is the attempt to write a somewhat coherent description (at least how I see it ;-)) of the results from discussing on the mailing list.

The current behavior is that a dummy device "usb-serial" is used. When this device is selected, the device driver is using a (VID/PID-Based) detection mechanism to attach a Chip-specific driver-implementation to the first found USB-to-Serial interface. This works fine as long as there is only one USB-to-Serial interface attached and the VID/PID is known.

Two scenarios are not working with the current implementation:
- More than one known USB-to-Serial interface is attached
- A USB-to-Serial interface without known VID/PID pair, but which is supported by usb-serial-for-android is used.
Both this scenarios are solved with my proposal.

User-Facing change 1: Listing actual USB devices.
- Instead of populating the device-list with a dummy "usb-serial" interface, actually display all connected USB-Devices (without knowing if they are USB-to-Serial interfaces) in the device-list. Something like:
  - "USB Device: CP2102 USB to UART Bridge Controller"
  - "USB Device: FT230X Basic UART"
- The user can then select which USB device he/she wants to use.

User-Facing change 2: Allow for Driver selection.
- Instead of populating the device-list with a dummy "usb-serial" interface, actually display all connected USB-Devices (without knowing if they are USB-to-Serial interfaces) and all drivers known to usb-serial-for-android in the device-list. Something like:
  - "USB Device: CP2102 USB to UART Bridge Controller (autoselect driver)" // This would be the same behavior as in usecase 1
  - "USB Device: CP2102 USB to UART Bridge Controller (autoselect driver)"
  - "USB Device: CP2102 USB to UART Bridge Controller (autoselect driver)"
  - "USB Device: FT230X Basic UART"
- The user can then select which USB device he/she wants to use.

### Operating system:

Android

### Subsurface version:
master, in the future?

### Steps to reproduce:

### Current behavior:
Dummy-Connection "usb-serial" is used. This automatically uses the first detected USB-Serial interface.

### Expected behavior:
Different USB devices are displayed with a description. If the respective setting is activated, a specific serial-driver can be used for the selected device.

### Additional information:

### Mentions:
@dirkhh

dirkhh, 2020-03-09:
The nice thing about using 'usb-serial' is that the user doesn't need to know a thing about the cable that they use. Which is true for (my estimate) 95+% of all divers. I **_really_** care about this stuff. And I always need to look up which chip the Mares Icon HD uses.
Additionally, I would guess that in 99.9% of all cases, people will have only one USB device attached. I mean... how DO you attach two dive computers at the same time, just so I can test this if we were to implement it? I guess somehow with a hub? Who does that???

Leaving that aside, as we discussed, there is that case of a different cable that we haven't seen, but that could be used if we allowed such a mechanism. That use case seems perfectly valid.

I will ponder this a bit more - it's not that I dislike your idea, @charno , it's that I want to make sure we make things easy for the common case. And _then_ figure out how to make the complex case possible.

charno, 2020-03-09:
Hi Dirk,

Thanks for reviewing. First let me say that this Idea is not important for me - it's important that I now can download dives from my computer to Android :) And you are for sure right that this works as intended for 99.9% of our users!

Hmm... Just thinking loud: 
- Maybe we could keep the UI like it currently is for the "normal" use case (One USB device only) with some human-readable string like "usb-serial". In the background, the (only existing) UsbDevice-Object could still somehow be attached to the list item and passed to the serial_android_usb_open function.
- If more than one USB device is present, display them separately and with device description in the list. Pass the UsbDevice-Object of the selected entry to the serial_android_usb_open function.
- If the advanced-setting is activated, always display all USB-Devices with Device description (even if only one is present, maybe even PID/VID for easier reporting of successes?), and for each device also present the option to use a specific usb-serial-for-android device class.

dirkhh, 2020-03-10:
How would we pass around the USB device?
I assume `serial_usb_android_open()` would somehow need to pass this as an argument to the JNI method - but I'm not sure how we'd do that. What's the natural way to do this?
Reading the Java code (which I'm not really familiar with), it seems like we create a prober that uses VID/PID to pick a driver and then simply get the first port of that driver - and you want us to be able to simply pick a specific driver and for that driver pick a specific port... but then how do we pass this information around?

I'm willing to work on all of the UI parts of this... but I'm bring really slow to wrap my mind around the question of "what do we get from Android, what do we pass around in C/C++ (and somehow through the QML UI), and what do we pass back into Android.

Right now your text above seems to assume that there are 5 drivers - are there always these 5? Could there be fewer or more? Can we get that list from Android?

The Java code, however, appears to only refer to 3 of those 5. I can't see references to Prolific, FTDI, and CdcAcm - but no reference to the SiLabs driver (and I don't know what the CH34xx is that you also mention above).

So I'm confused how this works today - and I'm not sure how to implement the behavior you are asking for.

Can you stub out an `..open()` function that passes information into the Java code that would allow you to simply open a specific usb device and use a specific driver to do so? And a probe function in Java (and related C++/JNI glue) that would return to us the information in that format so we can present it to the user? I.e., can you show me the API in C++ that I would need to interact with in order for this to work?

charno, 2020-03-10:
> How would we pass around the USB device?
> I assume `serial_usb_android_open()` would somehow need to pass this as an argument to the JNI method - but I'm not sure how we'd do that. What's the natural way to do this?

Passing the `QAndroidJniObject` (if using the https://doc.qt.io/qt-5/qandroidjniobject.html#QAndroidJniObject JNI abstraction, which Anton showed me and made my live much easier!) that represents the Java `UsbDevice`.

> Reading the Java code (which I'm not really familiar with), it seems like we create a prober that uses VID/PID to pick a driver and then simply get the first port of that driver [...]

That's correct for the current implementation. This needs to be changed depending on the parameters that we get, that's something I can do.

> [...] and you want us to be able to simply pick a specific driver and for that driver pick a specific port... but then how do we pass this information around?

When the USB devices get enumerated (I proposed in `set_non_bt_addresses`, but could be anywhere), you will get a `QAndroidJniObject` representing the Java `UsbDevice`. This device would have to be attached to an entry of the `connectionListModel` (which currently only contains strings). When pressing the Download-Button, the QAndroidJniObject would then have to be passed to the `serial_usb_android_open` function as an additional parameter.
 
> I'm willing to work on all of the UI parts of this... but I'm bring really slow to wrap my mind around the question of "what do we get from Android, what do we pass around in C/C++ (and somehow through the QML UI), and what do we pass back into Android.

From Android we get the `UsbDevice` (when calling via Qt JNI as `QAndroidJniObject`) when enumerating the USB-Devices using `UsbManager.getDeviceList()`.
The List of possible device drivers is static.

The `UsbDevice` and (if the setting for "Advanced device selection" is active) the name of the driver need to be attached to the entries of the `connectionListModel` and passed around.

To `serial_usb_android_open` the `QAndroidJniObject` representing the `UsbDevice`, and (if the setting for "Advanced device selection" is active) the name of the driver needs to be passed. From there I can take over and implement the passing to Android.

> 
> Right now your text above seems to assume that there are 5 drivers - are there always these 5? Could there be fewer or more? Can we get that list from Android?

Currently the usb-serial-for-android implementation provides these 5 drivers. From a very quick view I couldn't find an obvious method to list them in the usb-serial-for-android source. Since the driver classes only change when usb-serial-for-android is updated, I think a static list would be sufficient.

> 
> The Java code, however, appears to only refer to 3 of those 5. I can't see references to Prolific, FTDI, and CdcAcm - but no reference to the SiLabs driver (and I don't know what the CH34xx is that you also mention above).

In the usb-serial-for-android provided `getDefaultProbeTable()` method, the probetable gets already populated with the PID/VID/Class throuples that are known to usb-serial-for-android (only the official PID/VIDs, as far as I understand). Then later we add the ones that are not present in the table by default, but known to be used by dive computer cables.

CH34xx is another USB-to-Serial chipset that's supported by usb-serial-for-android, which until now doesn't seem to be used by any dive computer cables. But it might be in the future? That's exactly the moment the driver-selection feature comes in handy (known computer, unknown cable).

> So I'm confused how this works today - and I'm not sure how to implement the behavior you are asking for.
>
> Can you stub out an `..open()` function that passes information into the Java code that [...]

I don't know what you mean by stub out, but I can show you some possible C++ API:
```
/* for connecting */
dc_status_t serial_usb_android_open(dc_iostream_t **iostream, dc_context_t *context,
                                                            QAndroidJniObject usbDevice /* the UsbDevice */, 
                                                            std::string driverName /* Name of the driver class. If empty, the probetable is used to find a driver by PID/VID */)

/* The info that needs to be passed around. */ 
struct android_usb_serial_device_descriptor {
    QAndroidJniObject usbDevice /* the UsbDevice */,
    std::string className /* the driver class name. If empty, then "autodetect" */,
    std::string uiRepresentation /* The string that can be used for the user interface. */
}

/* get the default list of devices*/
std::vector<android_usb_serial_device_descriptor> 
serial_usb_android_get_devices();

/* get the list of devices including driver selection*/ 
std::vector<android_usb_serial_device_descriptor> 
serial_usb_android_get_devices_with_driver_selection()
```
Just because it's maybe easier than verbal description, below you see some dummy code for the later two methods:
```
std::vector<android_usb_serial_device_descriptor> 
serial_usb_android_get_devices()
{
    // Dummycode, including JNI calls:
    auto listOfUsbDevices = UsbManager.getDeviceList();
    // Special case to keep a generic user-facing name if only one device is present.
    if (listOfUsbDevices.size == 1) { 
        return std::vector<...> {android_usb_serial_device_descriptor {listOfUsbDevices[0], "", "usb-serial"} };
[... 2 lines cut ...]
[... cut ...]
    // Dummycode, including JNI calls:
    auto listOfUsbDevices = UsbManager.getDeviceList();
    std::vector<...> retval;
    for (auto usbDevice : listOfUsbDevices) {
        for (auto drivername : { "CdcAcmSerialDriver", "Ch34xSerialDriver", "Cp21xxSerialDriver", "FtdiSerialDriver", "ProlificSerialDriver" }) {
            retval.push_back( {usbDevice, drivername, usbDevice.getDeviceName() + " (" + drivername + ")" } );
        }
    }
    return retval;
}

```

What do you think? Does this clarify things for you?

dirkhh, 2020-03-10:
Yes, this really helps. I will try to get a first attempt to do this out in the next few days... I have quite a few things on my todo list right now, though.

dirkhh, 2020-03-10:
More reading, more thinking... I know I'm slow wrapping my mind around this.
Could you create a PR that  provides actual implementations of those last two functions, i.e. the `_get_devices()` functions, as well as the `serial_usb_android_open()` function that takes those extra two arguments?

Once I see how those work, passing things around and making them accessible from the UI is something I believe I can do reasonably quickly.

dirkhh, 2020-03-10:
more explanation - now that you have shared the dummy code I finally think I understand how this would work and agree with you that this is a good approach.

glance-, 2020-03-10:
@charno I really like this idea and it feels like you did a great job convincing @dirkhh to, so I'm happily watching this from the side-lines.

charno, 2020-03-10:
@glance- Thanks, it took long enough ;-) But that's more a testimonial to my explanation skills than to Dirks understanding.
@dirkhh I hope I will find some time to implement this in the next days. Holidays are over :-( 

Currently I'm going crazy trying to implement timestamps in the libdivecomputer.log, Time with sub-second resolution in portable C99 is reeeeealy hard it turns out.

dirkhh, 2020-03-10:
long enough? Oh Christof... you have no idea. This was really swift.
Convincing me of an idea that I don't like... I think the dive site management idea was being discussed for something like 3 years before it ended up going in... and even then I endlessly complained.

As for the timestamps... the code to do this exists. Yeah, maybe it's a private API in libdivecomputer... so just copy the code. The licenses are compatible.

charno, 2020-03-11:
Hi,

So I found the time on this bug. The result can be found at https://github.com/Subsurface-divelog/subsurface/compare/master...charno:usb-serial-new-interface.

There is only one commit, since all the changes are really interconnected (besides one comment, and a commit for one small comment wasn't worth it I think...).

What's implemented:
- A data structure that contains all the data that can be used to
describe an usb device (including user-facing string).
- A function to get a list of all attached usb devices (optionally with
selectable driver class).
- Changes in the serial_android_usb_open-function and in the Java part
to use the information about the usb device and optionally selected
driver when connecting.

This commit keeps compatibility with the current UI-Code in the case
that only one USB-Device is connected. If two devices are connected,
only the first one is tried.

There are still some small things to do:
- Change the user-facing string to something more descriptive.
- Parts which aren't uesd anymore when the UI-Part is implemented are
simply marked as obsolete (to keep compatibility for now).

But generally it seems to work. I could not test for multiple USB devices, but from the code I'm optimistic that it also works.

I did not yet open a pull request, since I'm not sure what workflow makes sense now that @dirkhh takes over for the UI part. If it's easier to open a pull-request, I can do it tomorrow, or if you want to do it feel free.

charno, 2020-03-11:
Ah, I've just seen that you actually asked for a pull request, so I've done one: https://github.com/Subsurface-divelog/subsurface/pull/2668

dirkhh, 2020-03-13:
Quick preview of the UI with two computers connected via a hub
[image]

dirkhh, 2020-03-13:
I'm thinking I should drop `/dev/bus` - but even with this the experience will still be challenging on smaller screens.
On the flip side, in non-override mode, having just
`/usb/001/004`
`/usb/001/003`
as the two entries is of course perfectly fine...

charno, 2020-03-13:
Nice! Love to see the progress :)

I think in the "normal" usecase the list would be reasonable, besides the problem that we need to somehow get better names for the devices.

charno, 2020-03-13:
Ah yes. I can try to update my side of the code to get better device naming. I'm not quite sure what a good naming concept would be.

What could be done for sure would be to parse the path, since the only valuable information in it are the bus- and device-number.

Also, android provides fields for device manufacturer name and device name, so I could try to incorporate these, but I fear that these will lead to very long strings.

I try to think of something reasonable. If somebody has a good idea, let me know :)

dirkhh, 2020-03-13:
For the cases where we know how to deduce the dive computer name from the USB information, getting a better device name is of course fairly easy. For the other cases, I'm not sure what a better name would look like. I guess we could at least translate the VID into a chipset brand or something... what I have done for the intent is that I just concatenated the whole string so that the C++ code can do the parsing and decide what to pass to the UI...
I guess for the comfortable Java programmer this will sound like a weird choice... but for me that was much more natural.
The manufacturer and product names often are useless (OSTC being the exception). E.g., for the Mares Icon HD this reads "SERUSB"/"USBSerial" - which is less than helpful. But because of the custom VID/PID I of course can easily turn this into Mares Icon HD...

So... it seems to make sense to turn the long-ish device string into something as I indicated above (`/usb/bus#/ep#`) plus the detected vendor (from the VID). Agreed?
Would you prefer to do that in Java or in your C++/JNI glue code? Or do you want to replicate what I did in the intent and pass the whole concatenated string to my code?

charno, 2020-03-13:
Hmm...

Generally speaking I would prefer to do this in the C++/JNI glue code.

Looking at the code in `QMLManager::showDownloadPage` I see that there are 3 exact models and 3 vendors that can be determined. There seem to be quite a lot of "generic" PID/VIDs. So I think replicating (or reworking) this for the edge-cases is not reasonable, especially since one of the edge-cases IS an unknown PID/VID pair.

What do you think of this?:
`<vendor as reported by android, shortened to 16 (?) characters> [<bus# as integer>:<dev# as integer>]`

For the two usb-to-serial converters I have here, this would give:
```
FTDI [1:2]
Silicon Labs [1:4]
```

After the change there will be only USB devices and BT devices in the list, which already have very different formats, so I think some sort of "usb" marker can be omitted.

dirkhh, 2020-03-13:
I'm ok with that.
I am completely rewriting the `showDownloadPage()`, so there is no need to try and be compatible with the code that's there. I will also rip out most of the intent code as there's no point in parsing all this twice - we'll use the intent in order to trigger showing the download page, but we'll get the information from your infrastructure.
There is a second set of changes that ensures that we also query your infrastructure when the user manually enters the download page (i.e. if they clicked 'no' when Android asked to open Subsurface-mobile)

charno, 2020-03-13:
What's your plan for getting the DC type/vendor? Updating the field when an item from the connection list is selected? (If BT computers have defined adresses this might also work for them?)
Should I also include the `UsbDevice.toString()` that you are currently using in the `showDownloadPage` function to the `android_usb_serial_device_descriptor`? Then all the JNI code for android-serial is contained in `serial_usb_android.cpp`.

Please also query the usb devices when the user klicks the refresh-button in the download page.
 
One more thing that would be really nice: When the usb-device is used for the first time in the divecomputer-dialog (and the USB_DEVICE_ATTACHED intent was NOT used), I ask for permission. If possible, could you add a receiver for the Intent "org.subsurfacedivelog.mobile.USB_PERMISSION", and simply start the download again? 
There is some exmample code at https://developer.android.com/guide/topics/connectivity/usb/host#permission-d for this.

dirkhh, 2020-03-13:
BT is completely different. The address tells us nothing, but many have a name that we can make sense of. And of course, many of those names are bogus and misleading. Several models sharing the same names, names changing with different firmware versions. Names including serial numbers. Nonsensical names (four hex digits that represent the internal hardware model code - who the heck thought that was a good idea?).

Right now what I am doing  for USB serial on Android (as you saw) is to parse the full string - which is total overkill as really only a couple of the fields hold useful information. vid/pid and (very rarely) manufacturer and product.

I will definitely query the USB code when the use clicks re-scan (or whatever that button is called).

I'll look into the permission thing - that's all "Böhmische Dörfer" for me...

(sorry to be so slow in responding - this is the middle of the workday for me and things are rather busy at my day job these days)

charno, 2020-03-13:
Don't worry about response time, I think it's great to get a response in one day for a project you do in your spare time (I assume?)! It's stressful and confusing times for everyone...

I can do the android/Java-part of the permission stuff. Can you do the qt/c++ part to "re-click the download button"? Maybe as a JNI-Method (similar to the `setDeviceString`-method, but probably without parameter)?

Should I give you these fields in the struct? pid / vid / manufacturer-name and device-name as separate fields?

dirkhh, 2020-03-13:
Yes, this is in my spare time. In my day job I'm an exec in a publicly traded software company :-)

It would be much easier to have these four as separate strings. So we'd get the full JNI UsbDevice object, the driver class name, the ui-string, pid, vid, manufacturer and product name. Then I can react to all the information we have.

I'm not 100% sure what you are asking me here:

> Can you do the qt/c++ part to "re-click the download button"? Maybe as a JNI-Method (similar to the setDeviceString-method, but probably without parameter)?

Are you asking to have something that the Java code can invoke that acts as if the user had clicked the download button? That's easy, I can certainly do that.

Feel free to create code that stubs out stuff that makes no sense to you, like
```
 // here the UI pops up the download window and parses this string

// this function needs to also do A) B) C)
```

charno, 2020-03-13:
> Yes, this is in my spare time. In my day job I'm an exec in a publicly traded software company :-)

I know who you are, I've seen Linus and you in 2018 (?) in Prague on stage. :-)
 
> It would be much easier to have these four as separate strings. So we'd get the full JNI UsbDevice object, the driver class name, the ui-string, pid, vid, manufacturer and product name. Then I can react to all the information we have.

Perfect, I hope I will find some time coding tomorrow.

> I'm not 100% sure what you are asking me here:
>
>
> Are you asking to have something that the Java code can invoke that acts as if the user had clicked the download button? That's easy, I can certainly [...]

That's exactly what I mean.

dirkhh, 2020-03-13:
Prague was 2011 and then again 2017... I've been to, err, a few of those events...

Having you show up and help us with the Android code is like a miracle.
Thank you so much for working on all this. It's really appreciated.

charno, 2020-03-14:
Ah yes, it was 2017. I really liked the interview/discussion format! 

Thanks for the warm words :) I'm happy to help, and also happy to go to dive trips without my laptop in the future.

I have now done all the discussed changes (including the wakelock as discussed on the mailing list). I did a force-push to the branch with the pull-request, so most likely you will have to rebase your work.

And, @dirkhh, could it be that you did not update the play store beta after the pull request https://github.com/Subsurface-divelog/subsurface/pull/2670 for the startup problem was merged? I still have this problem in the play store version... And I think that maybe Linus' problem with uploading dives might have the same source.

@torvalds could you test with your Pixel 4 and screen-timeout again with the CI build of the branch https://github.com/Subsurface-divelog/subsurface/pull/2668 ? Then I would know that the wakelock fixes that problem.

dirkhh, 2020-03-14:
> Ah yes, it was 2017. I really liked the interview/discussion format!

Me too. I have done a couple dozen of those with Linus over the years, and quite a few more with other colleagues. The conversational format without slides is more fun to do and overall seems to connect better with the audience.

> Thanks for the warm words :) I'm happy to help, and also happy to go to dive trips without my laptop in the future.

For me that's unlikely. Work as a maintainer seems to never stop - especially since I'm also the sole maintainer of all of the infrastructure...

> I have now done all the discussed changes (including the wakelock as discussed on the mailing list). I did a force-push to the branch with the pull-request, so most likely you will have to rebase your work.

Of course, that was expected :-)

> And, @dirkhh, could it be that you did not update the play store beta after the pull request #2670 for the startup problem was merged? I still have this problem in the play store version... And I think that maybe Linus' problem with uploading dives might have the same source.

I updated the beta about 90 minutes ago. Forgot to do so yesterday. See the comments about being the sole maintainer above. :-)

> @torvalds could you test with your Pixel 4 and screen-timeout again with the CI build of the branch #2668 ? Then I would know that the wakelock fixes that problem.

Linus doesn't get notifications from GitHub (WAY too many people @-tag him for that to be useful). Just send him a direct email and he is typically quite responsive :-)

dirkhh, 2020-03-21:
I believe we can consider this one closed.

closed 2020-03-21 by dirkhh