subsurface/subsurface #2665: usb-serial-for-android: Suunto computers

charno, 2020-03-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
With the new usb-serial-for-android implementation, we got feedback from quite some users that they had trouble downloading dive from suunto dive computers.

In this Issue I try to somewhat systematically collect feedback. 

**Any additional feedback would be much appreciated!** To get a version of Subsurface-Mobile with the new usb-serial-for-android support, you can enable the Beta channel for subsurface in the play store.

The following information would be helpful:
- Success? Partial success? Not connecting? Some other status?
- Exact model of the dive computer
- Information about the USB cable: PID / VID. To get this information, you can download the App "USB Device Info" from https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator. The important values are Product ID and Vendor ID
- libdivecomputer.log. One way to get it is to  menu -> "help" -> "ask for support". In the mail editor, copy all the lines from below "------- libdivecomputer.log --------".

### Operating system:
android

### Subsurface version:

At least until https://github.com/Subsurface-divelog/subsurface/commit/ca75177f0cc6d365c1c7af02a94732ddb9da8757

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

Current feedback overview:

| Status | Model  | DC driver | Cable VID/PID | Comments |
| ------------- | ------------- | ------------- | ------------- | ------------- |
| ✗ Gets stuck on "Connecting" | Suunto Vyper Air | suunto_vyper2.c | | "cheap" Inofficial cable used / did magically work later (maybe timeout fix?)
| ✗ Crashes after about 4 downloaded drives  | Suunto D4  | suunto_d9.c | 0403 / 6001 (FTDI) | did magically work later (maybe timeout fix?)
| ✗ Does not work. First read returns size=0  | Suunto Zoop  | suunto_vyper.c |  0403 / f680 (FTDI) |
| ✓ Report of working download  | Suunto D4i  | suunto_d9.c | 0403 / f680 (FTDI) | 

### Mentions:

dirkhh, 2020-03-11:
I just pushed a new beta - it will show up as 2.3.0 (4.9.3.1137).  And the APKs are available under downloads/test for sideloading.
That one has a hopefully very useful addition of high resolution timing in the libdivecomputer log file. In general, it would be good if anyone wanting to help out here could send an in-app support request after trying a Suunto download. Those emails go straight to me and I could extract both the VID/PID as well as some timing information from that. Of course, please include in the message body whether this did in fact download all the dives that you expected it to download.
@mturkia @charno @StephenGoodall

StephenGoodall, 2020-03-11:
@dirkhh I won't send the support request through and clog your inbox, because it is now working - if you do want it for successful attempts then let me know and i'll send both through.

I think part of my issue (Suunto Zoop - 0403/f680 - Official Cable) was the rubbish connector being too picky 😄 - once i pushed it at a different angle it was fine.

The timestamps do show up in the libdivecomputer logs now so they will be helpful for future cases i'm sure.

Cressi Leonardo (0403/87d0 - Unofficial - DiveMate IR adapter, I think this just emulates the proper device ID's as you can set the device in their app and it changes the VID/PID accordingly) worked aswell, again that was a bit picky but I think that was more of a physical interface issue rather than software.

Thanks to you (ad the core team) for continuing to push and improve Subsurface and @charno for getting the USB working on Android again, I think a lot of users have been waiting for this for a while now!

charno, 2020-03-11:
Thanks, @StephenGoodall 

I was also waiting for the USB download for a while, that's why I decided to give it a try :)

So the current status doesn't look so bad as it seemed in the mails. In fact, for all the computers that we have a report of failure we also got a report of a working download when the people tried later (maybe this was the timeout-fix that reached some people a bit later?)

dirkhh, 2020-12-12:
I can download from three different Suunto dive computers on Android, both with an original Suunto cable and with an "eBay" cable. Closing this.

closed 2020-12-12 by dirkhh