subsurface/subsurface #2676: Copy-paste of cylinder corrupts data

mturkia, 2020-03-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Seems that copy-paste of cylinder information is copying the O2%, start and end pressures also. This corrupts the data that is read from DC, i.e. pressures reported by DC.

### Operating system:

Linux

### Subsurface version:

Subsurface v4.9.3-1050-gb15b9c6cd0e8,

### Steps to reproduce:

- Copy cylinder information from one dive (ctrl-c)
- Paste the cylinder info (ctrl-v)
- Observe what all information is changed on equipment tab

### Current behavior:

O2%, start, and end pressures are changed.

### Expected behavior:

O2%, start and end pressure remain the same, only Type, size and working pressure are changed.

### Additional information:

### Mentions:

dirkhh, 2020-04-07:
The semantics here are interesting.
I think I understand the logic that @mturkia is proposing here. It seems reasonably intuitive.
Which means that the target dive should now have a least `source->numCylinders` tanks, and the first `source->numCylinders` should match the type/size/workingpressure` of those tanks in the source dive.

bstoeger, 2020-04-08:
What about the following use-case:
1) Add a number of cylinders to a (planned/manually created?) dive.
2) Use copy & paste to replicate those cylinders.

That also appears intuitive to me. However, I don't use multiple cylinders, so perhaps I'm mistaken.

What we could do if source has more cylinders than destination, is: copy over the existing cylinders and add the rest. Then the question arises: when adding the rest, I reckon we *should* copy %O2, %H2 but not start and end pressure?

Perhaps we should be more detailed when asking the user which data to copy/paste: Let them chose between cylinder type, gasmix and/or pressures...?

dirkhh, 2020-04-08:
I don't know if I want to make the UI more complex. It's already rather confusing.
I think the following logic makes sense:

- if there is no target cylinder (i.e., there are more cylinders in the source) then copy description, size, working pressure, gasmix
- if there is a target cylinder, copy description, size and working pressure, but NOT gasmix

There is one corner case that we have thanks to Linus making a very strange design decision that I argued against... there is no way to indicate "the gas mix isn't set", because 0/0 is "special" and indicates "air". I think one could make the argument that if the target cylinder has gas mix 0/0 then yes, we should overwrite that as well. But I can think of a situation where this isn't what you want - the most basic case. You want to set the tank in all of your dives and you happen to copy from a dive that has an EAN gas mix and now all of your dives that were on air suddenly have the wrong gas mix.
So yeah, maybe we do need additional UI for this :-( :-( :-(

bstoeger, 2020-04-08:
We could also differentiate by whether `manually_added` is set.

mturkia, 2020-04-10:
I agree with Dirk's comments. I would not personally copy the gas mix as that comes from my DC. But then again, I only care about single cylinder dives :D

The commit message of the fix sounds like it is even better than I asked for :D

PR #2734 (2020-04-10): undo: refine pasting of cylinders
When pasting cylinders, the destination dive got a verbatim copy
of the cylinders of the source dive. This is not what users want:
for example, the start and stop pressures from the dive computer
as well as the gas mix may be overwritten.

There seems to be no perfect solution, since some times users may
want to paste the gas-mix.

Therefore, let's choose a heuristic approach for now (in the future
we might implement a UI-flag):

When copying over existing cylinders, only copy type and maximum
pressure.

When adding new cylinders (i.e. from-dive has more cylin [... cut]
commit 4489389a (2020-04-10): undo: refine pasting of cylinders

When pasting cylinders, the destination dive got a verbatim copy
of the cylinders of the source dive. This is not what users want:
for example, the start and stop pressures from the dive computer
as well as the gas mix may be overwritten.

There seems to be no perfect solution, since some times users may
want to paste the gas-mix.

Therefore, let's choose a heuristic approach for now (in the future
we might implement a UI-flag):

When copying over existing cylinders, only copy type and maximum
pressure.

When adding new cylinders (i.e. from-dive has more cyli [... cut]
closed 2020-04-10 by dirkhh