subsurface/subsurface #2686: Android: Small UI issues

charno, 2020-03-20:
Finally found time to test the current mobile beta. I found some small UI nuisances, probably not blockers but probably also very easy to fix.

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

1)
When switching storage (Cloud -> local, local -> local was tested), the buttons in the divelist are strangely positioned.
[image]

2)
When a divecomputer is disconnected and reconnected (with the DEVICE_CONNECTED intent) while the download page is open, the connection list is inconsistent. The Connection from the intent is used, but when I click on the device list the old connection (bus/device id) is shown.
[image]

3)
The save button in the dive editing dialog should probably be white (as it is on the blue background on the divelist)
[image]

### Operating system:
Android 10 on galaxy s10

### Subsurface version:
3.0.0(4.9.3.1186) from Play store beta

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-03-20:
I have seen the first and have tried to debug it... but haven't been able to figure out _why_ the buttons shift right. Their position is based on the width of the page - so if that was wrong, then the whole page layout should be wrong, but it isn't (at least in my experiments).

The little screen video is a bit confusing to me, I'm not quite sure where it starts and what exactly you are showing there. Am I understanding you correctly that you get an incorrect connection listed? Does the download work?
I have seen a situation where a device was listed twice (once with an incorrect bus address), but tapping on rescan / Neu scannen fixed that in my case. Does it fix the situation for you?

Yes, the incorrectly colored action button is on my "to be fixed before release" list.

Thank you so much for this report - incredibly helpful to see what is not working for our users.

charno, 2020-03-20:
Hi Dirk,

Downloading does work, as said it's just a little UI annoyance.

Steps:
- Start subsurface with computer already attached to phone
- Go to download screen (btw., for me the only existing entry was not selected...)
- Disconnect the dc
- Connect the dc
- select "Subsurface-mobile" as the app to handle the attached USB device

Now the status is this:
- In the connection field, the computer is displayed with [1:3] (the new device number)
- When i press on the connection field, the list is displayed with only one entry (correctly so) with [1:2] (the old device number)
- Funny enough, when I press on the entry with [1:2] in the list, the field value does not change (still stays at [1:3])

Probably it's something like not refreshing the list when the USB device is passed in by DEVICE_CONNECTED intent.

dirkhh, 2020-03-20:
So I'm trying to recreate this behavior. And I'm not succeeding.
One of your comments I do see. If I'm already on the download page and then plug in a device, its connection isn't selected.
But I cannot reproduce the scenario where the bus number changes. I know I have seen this in the past, but I can't remember with which of my (way too many) Android devices... and with the ones that I tried right now, I can't.

charno, 2020-03-23:
Sorry for being a bit slow at the moment. Times are crazy.

I did just test the last CI build from #2687.

I noticed that there is (at least on my device) a problem when I plug in the dive computer and subsurface is already running (foreground / background does not matter). When I do this, and then press the Use Subsurface-mobile option in the popup, the app freezes for me. I had this behaviour in a earlier build as well, but not in the latest beta from Play Store (3.0.0(4.9.3.1186)).

I'm currently not on my private dev computer. If you want I can give you some adb logs later today or tomorrow.

dirkhh, 2020-03-23:
I cannot reproduce this.
I tried about ten different combinations of the state of Subsurface-mobile when I plugged in the device, and in as long as I am not in the middle of editing a dive, I always end up with a working app and on the download page. And with the correct connection selected.
Admittedly, I am not testing with the CI builds but with my local builds, so that could be the difference - but I believe I use identical Docker images to do the build...

mikeller, 2026-10-17:
Hi @charno,

Same situation as the related report (#2753) from the same era: this is now over 6 years old, and the mobile UI has been substantially reworked since, with several small-UI issues from this period already addressed. With no specifics here on what's still outstanding and no further reports, we're closing this as stale.

Please reopen with current screenshots and your Android/Subsurface version if you're still seeing specific issues on a current build.

closed 2026-10-17 by mikeller