subsurface/subsurface #2714: Mobile dark theme improvement

willemferguson, 2020-03-31:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

The **Dark** colour theme on mobile looks extremely inconvenient on my phone and I am sure on other phones as well. The main reason is that the colour of the dive trip headers is a very light blue. Firstly this negatively affects the readability of the dive trip information and, secondly, it does not fit with a dark colour theme. It just does not contribute to my user experience. I propose that the colour of the dive trip headers be made a much darker blue that harmonises with a dark theme. It could even be made a pretty dark navy colour, with the year/month indicator a bit lighter in colour.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh

dirkhh, 2020-03-31:
I never understood dark themes.
I distinctly remember about a week ago, while waiting for people to test the beta, that I was going to look at the utter mess of that and then I found something more interesting to do and forgot again.
Yes, it's completely unusable and an embarrassment.

My first inclination would be "I don't use it, no one seems to care enough to fix it, so remove it".
The pink theme is hideous, but has one enthusiastic user. In its current state I cannot imagine that anyone is using the dark theme.

bstoeger, 2020-04-07:
The pink theme is so bad that it's good. I would call it "Pinkie Pie", but that might create copyright issues.

Fixing the dark theme might be one of the things people not proficient in coding could contribute.

dirkhh, 2020-04-07:
The pink theme is the theme that Linus uses. I'm not joking. Actually, it came into existence as a joke of mine, and Linus loved it and asked me to make it a feature :-(
Yes, it's HORRIBLE.

mikeller, 2026-10-18:
Thanks for the report, and apologies that this sat open for so long.

Looking at the current source, the trip header colour issue described here was
addressed in commit 7ed861fb9 ("mobile/UI: better dive list color for dark
theme", 18 Dec 2020). It changed the dark theme's trip header background from
the light blue originally reported to a dark navy (#303050), with the
year/month date badge using a lighter indigo accent (#3F51B5) — which matches
what was suggested in the original description. This value is still in place
on current master.

I've only verified this by reading the source and computing text contrast
(trip title text on the new header background comes out at roughly 11.9:1,
well above normal readability thresholds), not by running a current build on
a device. So there's one thing I can't rule out from source alone: the new
header background (#303050) is quite close in brightness to the plain dive
list background (#303030), which could still make the header look like it
blends into the list even if the text itself is easy to read.

Could someone who still sees this on a current build confirm whether it now
looks right, or whether the header still blends in or otherwise looks off? If
it looks fine, we'll close this as fixed by 7ed861fb9. If something still
looks wrong, please let us know exactly what (e.g. blending into the
background vs. an actual light-blue colour) so we can take a closer look.