subsurface/subsurface #2718: Subsurface Android 3.0.2 Dives not uploading to cloud.

Jeffklop, 2020-04-02:
Version 3.0.2 of Subsurface for Android will not upload new dive to the cloud.

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The Android 3.0.2 will read the cloud for any updates made by using the Windows 10 version of Subsurface, but not write to the cloud.  
### Operating system:

WIndows 10 Home
English
Inspiron 13-7353

### Subsurface version:

Android 3.0.2
Windows 10 version works correctly
Official release

### Steps to reproduce:

1. Download new dives from Shearwater Perdex AI to Android Subsurface 3.0.2
2. Accept dives, to enter into local cache of Android Subsurface 3.0.2
3. Select "Manual Sync with Cloud" on Android Subsurface 3.0.2
4. Wait for cloud update.
5. Open Windows 10 Subsurface 4.9.3.
6. The new dives are not displayed.
7. On Windows 10 Subsurface 4.9.3, select "Open Cloud Storage"
8. The new dives are still not displayed.
9. Reopen Android Subsurface 3.0.2
10. The new dives are not displayed on the Android Subsurface 3.0.2.

Note: If I make a change in the Windows Subsurface and save to the cloud and I load the cloud on the Android Subsurface 3.0.2 the change appears. It looks like the Android Subsurface 3.0.2 can read the cloud but just does not write to the cloud.

Previous version of Android Subsurface worked.

### Current behavior:

Android Subsurface 3.0.2 does not save changes to the cloud. 

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-04-02:
That is indeed a very bad bug. I'll fix this today and push out a new version.

closed 2020-04-04 by dirkhh

Jeffklop, 2020-04-05:
> <https://github.com/Subsurface-divelog/subsurface/issues/2718> via 0e1b784

Is there a new Android version?

dirkhh, 2020-04-05:
A new beta for 3.0.3 should be in the beta channel on Google Play in the next ten minutes to twelve hours, depending on, err, something Google something.
That should hopefully fix the majority of currently open issues.
Once a few people have tested and reported back, I'll release that.

Binaries for side-loading are available https://subsurface-divelog.org/downloads/test/

Look for the ones marked 1328 (both 32 and 64bit versions are there)

Jeffklop, 2020-04-05:
The 3.0.3 Android version fixed the writing to the cloud problem. Thank you.

From: Dirk Hohndel
Sent: Saturday, April 4, 2020 10:33 PM
To: Subsurface-divelog/subsurface
Cc: Jeffklop; Author
Subject: Re: [Subsurface-divelog/subsurface] Subsurface Android 3.0.2 Divesnot uploading to cloud. (#2718)

A new beta for 3.0.3 should be in the beta channel on Google Play in the next ten minutes to twelve hours, depending on, err, something Google something.
That should hopefully fix the majority of currently open issues.
Once a few people have tested and reported back, I'll release that.

Binaries for side-loading are available https://subsurface-divelog.org/downloads/test/

Look for the ones marked 1328 (both 32 and 64bit versions are there)

commit b8d70590 (2020-04-02): mobile/core: remove locking for git access

We have convinced ourselves that only the main thread will ever trigger
a save operation, therefore the locking is not needed (and it has
recently started to cause user problems where local changes aren't saved
to storage and get lost).

Fixes #2718
commit 4a097915 (2020-04-02): mobile/core: remove locking for git access

We have convinced ourselves that only the main thread will ever trigger
a save operation, therefore the locking is not needed (and it has
recently started to cause user problems where local changes aren't saved
to storage and get lost).

Fixes #2718
commit 7cc14343 (2020-04-04): mobile/core: remove locking for git access

We have convinced ourselves that only the main thread will ever trigger
a save operation, therefore the locking is not needed (and it has
recently started to cause user problems where local changes aren't saved
to storage and get lost).

Fixes #2718
commit 0e1b784a (2020-04-04): mobile/core: remove locking for git access

We have convinced ourselves that only the main thread will ever trigger
a save operation, therefore the locking is not needed (and it has
recently started to cause user problems where local changes aren't saved
to storage and get lost).

Fixes #2718