subsurface/subsurface #2731: ensure mobile downloads get dive numbers

dirkhh, 2020-04-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

reported by @willemferguson 

### Subsurface version:

Subsurface-mobile 3.0.3

### Steps to reproduce:

download from dive computer, all dives have dive number 0

### Expected behavior:

consecutive number in line with the current numbering of the dive list (same as desktop)

mwerle, 2020-08-13:
This is not limited to Subsurface-mobile; I noticed the same happened using the desktop client (Windows 4.9.6).

It -may- be related to changing (primary) dive-computers.  I was importing from my Perdix AI, and then from my Cosmiq+ and everything was working as expected. Currently my Perdix is away for repairs, so I have had to perform the import from the Cosmiq which has resulted in dives being numbered '0'.

dirkhh, 2020-08-13:
This is so odd. I am looking at the code right now. If the dives come from the dive computer with no number (which they should) and if the last / newest dive in your dive list has a number, then the code should number the new dives (that are newer than that last dive) with consecutive numbers after that. And I did a download from my dive computer just last week and this worked as intended and my dives got the correct number.
So I am trying to understand what's different for me. 
I posted that bug after Willem mentioned this as a problem in an unrelated thread, but I still haven't managed to actually reproduce this. If you could try to reproduce this and show me how you reproduced it, I'd love to see a screen capture video or something because I'd love to track this one down... but just by looking at the code I can't quite figure out what's going wrong.

mwerle, 2020-08-14:
Will do a couple more dives this weekend. If I can repro it I will run the import through the debugger and/or supply sample data.

mwerle, 2020-08-17:
Ok, found it ( at least in my dataset )

divelist.c:1274 :

```
	if (!sequence_changed && nr >= preexisting && !new_dive_has_number) {
		for (i = start_renumbering_at; i < dives_to_add->nr; i++)
			dives_to_add->dives[i]->number = ++nr;
	}
```

sequence_changed == 0
nr == 287 <-- this is the number of the latest dive in my list
preexisting == 288  <-- this value is the total number of dives in the list
new_dive_has_number == 0

So the new dives never get renumbered because I manually modified the sequence numbers of dives at some point.  When I import using my Shearwater AI it "works" because the Shearwater provides dive numbers to Subsurface.

So question: _why_ does Subsurface guard total number vs latest number?  It should only worry about the latest number and continue counting from there; whatever the user has manually done to reset numbering at some point in the history should have no bearing on new imports.

dirkhh, 2020-08-17:
I think that's clearly a bug. I'm trying to go back in time to figure out what the thought process here might have been -- my vague recollection goes something like "we only do this if the dives are all already numbered" or some such logic (in which case this test presumably makes sense). Or, more likely, what we REALLY wanted to check was that the newest dive that was already there was indeed the one with the highest number? But then the check makes no sense at all.
What happens if you drop this check? My guess is it will work, correct?
I'll ask @torvalds to chime in as well...

mwerle, 2020-08-17:
Ah... I never changed the code to see.  I'll try tomorrow (knackered, sorry) but yes, my assumption is that it'll work.  Still, might be good to see if anybody can remember what it's guarding against just in case.

torvalds, 2020-08-17:
We used to just merge new dives with any preexisting ones, and after we had done that, we'd just look at the last dive we had *before* downloading, and do a simple

        /* If there are dives in the table, are they numbered */
        if (!last || last->number)
                try_to_renumber(last, preexisting);

ie the original logic was simply to say that if we had any number at all on the last dive (or no dives at all), then we'll continue from that point. 

But then that function was split up, and it became a lot more complex, and it tries to look if adding new dives happened before the last dive etc. That happened in a few steps, but the major change was in commit 0249e1258 ("Import: split process_imported_dives() function") and honestly, that 

    nr >= preexisting

test doesn't make any sense that I can tell. They are two completely different numbers: "nr" is from a "dive->nr", while "preexisting" is a count of dives (it comes from a dive _table_ nr).

Dive numbers (dive->nr) and dive table size (dive_table.nr) have nothing to do with each other. It's like comparing a length to a temperature. Different units.

So yes, I think that check should be removed. But the author is @bstoeger so maybe he intended something else.

Berthold?

bstoeger, 2020-08-17:
Clearly an editing error.

There is no point in remembering `preexisting`, since we don't modify `dive_table` anymore. After all, that was the whole point of the commit.

Judging by the comment above, the `nr = ...` line should probably read as
```
        bool last_dive_is_numbered = dive_table.nr > 0 ? dive_table.dives[dive_table.nr - 1]->number > 0;
```
and the `if ` should be changed to
```
        if (!sequence_changed && last_dive_is_numbered && !new_dive_has_number) {
````

I can do a PR if you want.

dirkhh, 2020-08-17:
Yes, @bstoeger - this sounds reasonable. Please do a PR.
Thanks, @torvalds for refreshing my memory how simple this used to be. I should have spent more time with `git blame` before tagging you :-)

commit c7b5ce40 (2020-08-17): core: fix dive renumbering logic on import

0249e12 split up the dive import logic in multiple steps. Thereby,
the one of the conditions for renumbering the imported dives (is
the last old dive numbered) got messed up: The first number of the
new dive was compared to the total number of old dives, which makes
no sense.

- Simply check for the number of the last existing dive (if any).
- Don't remember the number of old dives - the original table is
  not modified anyway.

Fixes #2731
PR #2915 (2020-08-17): core: fix dive renumbering logic on import
0249e12 split up the dive import logic in multiple steps. Thereby,
the one of the conditions for renumbering the imported dives (is
the last old dive numbered) got messed up: The first number of the
new dive was compared to the total number of old dives, which makes
no sense.

- Simply check for the number of the last existing dive (if any).
- Don't remember the number of old dives - the original table is
  not modified anyway.

Fixes #2731

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [  [... cut]
commit f24fe107 (2020-08-17): core: fix dive renumbering logic on import

0249e12 split up the dive import logic in multiple steps. Thereby,
the one of the conditions for renumbering the imported dives (is
the last old dive numbered) got messed up: The first number of the
new dive was compared to the total number of old dives, which makes
no sense.

- Simply check for the number of the last existing dive (if any).
- Don't remember the number of old dives - the original table is
  not modified anyway.

Fixes #2731
closed 2020-08-17 by dirkhh