subsurface/subsurface #2739: Petrel 2 not recognized as dive computer

ncgator82, 2020-04-10:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
 

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-04-10:
I'm puzzled that this wouldn't be recognized as dive computer...
```
	if (btName.startsWith("Predator") ||
	    btName.startsWith("Petrel") ||
```
Very strange.
There is a settings option to show all BT/BLE connections. If you enable that one (sorry, this is "per session only", so it does NOT stay enable between runs of the software), can you download from your Petrel2?
Please give that a try, and right after the download attempt (regardless of whether it's successful or not), please tap on Help->Ask for support in the app and send me the log that this will create as an email (the default recipient address that it will populate is just going to me, so this stays private in case you worry about that).

closed 2020-04-11 by dirkhh

ncgator82, 2020-04-11:
> Closed #2739 via 55bbdc2.

This release fixed my issue. Thanks so much!!

Rick Holcombe
Sent from my mobile phone, please pardon typos

commit 55bbdc27 (2020-04-11): BLE dive computer detection

Both Shearwater Petrel and Petrel 2 identify as 'Petrel' as their BT and BLE
names. But only the Petrel 2 supports BLE, thus only the Petrel 2 shows up in
the list of known dive computers on iOS (which supports only BLE but not
BT-only). By switching this around to always pick Petrel 2 we now correctly
detect such a dive computer on iOS.

Fixes #2739

Reported-by: Rick Holcombe <wrh@nc.rr.com>