subsurface/subsurface #2751: subsurface: relocation error

novoid, 2020-04-13:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

After upgrading Debian Stretch to Buster, subsurface installed via `http://ppa.launchpad.net/subsurface/subsurface/ubuntu` did not work any more.

### Issue long description:

Yesterday, I upgraded from Debian Stretch to Buster.  Subsurface did get uninstalled during that process.

I could not re-install using the previous settings:
```
root@host ~ # apt install subsurface          
Reading package lists... Done                                  
Building dependency tree                                                
Reading state information... Done                             
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.                        
The following information may help to resolve the situation:     
                                                      
The following packages have unmet dependencies:                         
 subsurface : Depends: libcurl3 (>= 7.18.2) but it is not installable
E: Unable to correct problems, you have held broken packages.
100 root@host ~ # 
```

I had to change my `/etc/apt/sources.list.d/subsurface.list` from
```
deb http://ppa.launchpad.net/subsurface/subsurface/ubuntu xenial main
```
to
```
deb http://ppa.launchpad.net/subsurface/subsurface/ubuntu bionic main
```

I then re-installed subsurface:
```
root@host ~ # apt install subsurface                                                                                                                                                                                                                                            
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libgrantlee-templates5 libqt5bluetooth5 libqt5bluetooth5-bin libqt5location5 libqt5location5-plugins libqt5positioning5-plugins libqt5positioningquick5 libqt5quickwidgets5 qml-module-qtlocation qml-module-qtpositioning                                                             
The following NEW packages will be installed:
  libgrantlee-templates5 libqt5bluetooth5 libqt5bluetooth5-bin libqt5location5 libqt5location5-plugins libqt5positioning5-plugins libqt5positioningquick5 libqt5quickwidgets5 qml-module-qtlocation qml-module-qtpositioning subsurface                                                  
0 upgraded, 11 newly installed, 0 to remove and 2 not upgraded.
Need to get 9,633 kB of archives.
After this operation, 27.2 MB of additional disk space will be used
```

After installing, subsurface did not start:

```
vk@host ~ % subsurface --verbose
Subsurface v4.9.3,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ce6d9896a79afaa82641132e338f8744714c8593)
built with Qt Version 5.9.5, runtime from Qt Version 5.11.3
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 3.0."
subsurface: relocation error: /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so: symbol _ZN11QGeoMapTypeC1ENS_8MapStyleERK7QStringS3_bbi10QByteArray version Qt_5_PRIVATE_API not defined in file libQt5Location.so.5 with link time reference
127 vk@host ~ % 
```

### Operating system:

Debian Buster (upgraded yesterday from Stretch, updated today). Using English as system language.

### Subsurface version:

- Subsurface v4.9.3 from 
```
deb http://ppa.launchpad.net/subsurface/subsurface/ubuntu bionic main
```

### Steps to reproduce:

Just try to start subsurface.

### Current behavior:

Error message as shown above.

### Expected behavior:

Subsurface is starting and showing existing data.

### Additional information:

### Mentions:

Issue #2257 seems to contain a similar error message.

dirkhh, 2020-04-13:
Subsurface needs private Qt header files and links to private Qt APIs that aren't stable between releases, often not even between minor releases. Is it possible that Buster is using a slightly different version of Qt than Bionic? The binary you are trying to install is linked against the (fairly old) Qt 5.9.5 that's still shipping with Bionic. Which Qt version is on your system?

novoid, 2020-04-13:
Hi @dirkhh,

I guess you can find the information necessary within the following snippet:

```
root@host /etc (git)-[master] # acsh libqt5core5a:amd64
Package: libqt5core5a
Source: qtbase-opensource-src
Version: 5.11.3+dfsg1-1+deb10u3
Installed-Size: 5528
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Architecture: amd64
Provides: qtbase-abi-5-11-3
Depends: libc6 (>= 2.28), libdouble-conversion1 (>= 2.0.0), libgcc1 (>= 1:3.4), libglib2.0-0 (>= 2.22.0), libicu63 (>= 63.1-1~), libpcre2-16-0 (>= 10.32), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4)
Suggests: libthai0
Breaks: libqt5scintilla2-12v5 (<< 2.9.2+dfsg-2~)
Description-en: Qt 5 core module
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 The QtCore module contains core non-GUI functionality.
Description-md5: 75c9109eafc0c1da6d6b6ca7b292f133
Homepage: http://qt-project.org/
Multi-Arch: same
Recommends: qttranslations5-l10n
Section: libs
Priority: optional
Filename: pool/updates/main/q/qtbase-opensource-src/libqt5core5a_5.11.3+dfsg1-1+deb10u3_amd64.deb
Size: 1942620
MD5sum: 4852bf7eebc7075db40e863b5e29c478
SHA256: bce769ae59d18a3a3905dd8f2c1ad3da4e2ae3a6dbac86d6f840640708c74ef0

Package: libqt5core5a
Source: qtbase-opensource-src
Version: 5.11.3+dfsg1-1+deb10u2
Installed-Size: 5528
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Architecture: amd64
Provides: qtbase-abi-5-11-3
Depends: libc6 (>= 2.28), libdouble-conversion1 (>= 2.0.0), libgcc1 (>= 1:3.4), libglib2.0-0 (>= 2.22.0), libicu63 (>= 63.1-1~), libpcre2-16-0 (>= 10.32), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4)
Recommends: qttranslations5-l10n
Suggests: libthai0
Breaks: libqt5scintilla2-12v5 (<< 2.9.2+dfsg-2~)
Description-en: Qt 5 core module
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 The QtCore module contains core non-GUI functionality.
Description-md5: 75c9109eafc0c1da6d6b6ca7b292f133
Multi-Arch: same
Homepage: http://qt-project.org/
Tag: role::shared-lib
Section: libs
Priority: optional
Filename: pool/main/q/qtbase-opensource-src/libqt5core5a_5.11.3+dfsg1-1+deb10u2_amd64.deb
Size: 1942300
MD5sum: 323df9f9d01e98a7982c0e7c7b636436
SHA256: f8fee14d330273e707397df662ad3bace45f5eac315b6b1259736d2705706b1f
```

I'm not a developer who knows C/C++ stuff. Does this mean that the only way I may get subsurface is by setting up the compile stuff, learning how to compile things and hope that there is no error message during that process I can not interpret? (I had my fair share of failing to compile kernels and so on twenty years ago.)

Debian stable (Buster) is far from being an exotic system. Therefore, I tend to think that there is a very high chance for a binary release of subsurface. ;-)

dirkhh, 2020-04-13:
Yes, the Qt 5.9.5 based binaries will probably fail on your 5.11.3 based distro.
As far as "relevance" goes... Linux is about 10% of our users, Debian is less than 10% of our Linux users. I haven't pulled stats for a while (and since those only get collected from people who opt in to check for updates, they may be skewed, anyway), but I'd say we are talking about maybe a few dozen Debian users out there, and not all of them on the same Debian version.
Yes, we should have a Debian build. I'll add it to my todo list.

novoid, 2020-04-13:
Wow, thanks for the insight. I'd have lost bets on that.
I'll follow your updates on the Debian binaries. Thanks!

(Not sure on the closing convention of the ticket in such a case.)

dirkhh, 2020-04-13:
Oh, I wouldn't have closed this - it's still outstanding and this helps to remind that I need to do this...

atdotde, 2020-04-13:
Maybe i am stating the obvious, but: habe you tried using the appimage? This is made for situations like this.

novoid, 2020-04-13:
> Maybe i am stating the obvious, but: habe you tried using the appimage? This is made for situations like this.

Hi @atdotde,

I just downloaded the AppImage beast (110MB!). After enabling the execute flag, I was able to run subsurface.

Until there is a proper Debian package (I prefer the clean, usual distribution-based approach), I'll use this as a workaround. Thanks for the pointer!

mikeller, 2026-10-17:
Hi @novoid,

Thanks for digging into the Qt ABI mismatch and confirming the AppImage worked as a workaround at the time. As you found, the PPA binary links against an older Qt than current Debian ships, and we need private, unstable Qt APIs that aren't ABI-stable even across minor Qt releases, a proper Debian package remains on our todo list but isn't likely to happen soon given how few users are affected.

One update since: these days we'd point you at our official Flatpak (https://flathub.org/apps/org.subsurface_divelog.Subsurface) rather than the AppImage as the better-maintained option on Linux outside your distro's own packages.

Closing this since a working alternative build resolved it for you. Please reopen if neither the Flatpak nor the AppImage works on your system.

closed 2026-10-17 by mikeller