subsurface/subsurface #2753: Android: small UI issues

janmulder, 2020-04-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Some small UI issues. Likely unrelated, but in one issue as there small.

1. After some editing in the divelist, like deleting dives, undo-ing, some of the thin lines between the individual dives in the list are gone, and I did not find any way to make them appear again, other than restarting the app.

2. The line on the bottom of a trip header in the divelist is much fatter than the line on the top of the dive trip header. Maybe intentionally (I don't know), but this looks strange and inconsistent.

3. Opening the "edit trip details" shows no keyboard on opening (which I think is just fine), but the center bottom (save) button icon is not drawn, and shows as a plain white circle. Opening the keyboard by clicking one of the editable fields, suddenly shows the save icon.

### Operating system:
 Android 9, LineageOS 16
 not relevant. LineageOS is 99% AOSP

### Subsurface version:
 4.9.3.1442

 Own android build, including 3 pull reguests merged in. Not relevant for this issue. 

### Steps to reproduce:
see above

### Current behavior:
Strange UI artifacts

### Expected behavior:
No strange UI artifacts

dirkhh, 2020-12-12:
1. the lines are always drawn, but QML sometimes seems to not render them. I have seen the issue as well, but have no idea how to fix it
2. this was at some point intentional, but I think the dive list overall is insanely ugly. I just have no idea how to make it less ugly. UI design isn't my strength, yet most of this stuff has been implemented (or changed) by me. With predictable results.
3. so I think this one I fixed a while ago.

Of course, if I ever get the current update of the mobile up to a reasonably current Kirigami and Qt 5.15 to work, hopefully we can focus on improving those UI issues again. Have you managed to build with the current build container? I think I have this working again... at least the GitHub Action completes (I am of course talking about the newKirigami branch).

janmulder, 2020-12-13:
"Have you managed to build with the current build container?"

No, unfortunately not. The problem is different than before, but equally mysterious. Compile of Kirigami fails and the primary complaint is "no ECM_DIR defined", and obviously, it is defined.

dirkhh, 2020-12-13:
The ECM_DIR thing has stumped me a few times. Can you post the script that you use?
I'm obviously really eager to make that easy for people.
I can try it here myself in a fresh VM - that should point me at the stumbling blocks.
The fact that the GitHub action succeeds makes me think that I have it all sorted out, but what matters is that it works for people like you...

mikeller, 2026-10-17:
Hi @janmulder,

This report is now over 6 years old, and from what we can tell several of the Android small-UI issues raised around this time have since been addressed as the mobile UI went through multiple subsequent rebuilds. Since there's no further detail here on what specifically remains outstanding, and no additional reports since, we're closing this as stale rather than leaving it open indefinitely.

Please reopen (ideally with current screenshots and your Android/Subsurface version) if you're still seeing specific UI issues on a current build, we'd genuinely like to know if something concrete remains.

closed 2026-10-17 by mikeller