subsurface/subsurface #2762: Depth field allows large values

jirkapok, 2020-04-15:
### Description
When entering depth application freezes. See the provided screenshot.
- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
The "Final depth" field in Dive planner points allows meaningless values to be entered.

### Operating system:
Windows 10

### Subsurface version:
Desktop Subsurface 4.9.3

### Steps to reproduce:
1. Create dive manually in planner
2. Enter values from screenshot in depth field fill "50000" meters
3. Application freezes and shows meaningless chart

### Current behavior:
Application freezes and shows meaning-less chart.
Not checked on another OS.

### Expected behavior:
Nothing is calculated, error popup is shown (based on other fields error behavior), chart isn't refreshed (or is empty).

### Additional information:
See screenshot with current behavior.
[image]

### Mentions:
N/A

atdotde, 2020-04-15:
We don't impose a boundary what we think is a "reasonable" depth and what not. We try to calculate a deco plan for anything possible, but if that is too long, we stop the calculation and show the error that you see in the red bar. For a dive to that depth, deco would take too long to be calculated. I don't see a bug.

jirkapok, 2020-04-15:
The application freezes and needs to be restarted. I think this should be enough to threat it as a bug, in any possible way. It doesn't have to be fixed by solution proposed by me.

atdotde, 2020-04-16:
For me, it does not freeze, it just gets very very slow (as it is trying to calculate a deco schedule that takes several days). It gets back to normal as soon as I replace the stupid depth with a reasonable one.

jirkapok, 2020-04-16:
Ok, leaving next steps up to you. I have no more comments.

commit eef0e546 (2020-10-03): planner: limit depth and time of segments to sane values

Using non-sensical depth and times for segments in the planner may
lead to an unresponsive UI. Therefore limit depth to 1000 m/3000 ft
and time to 100 h. Limiting of depth is done twice: once to a
value of 3000 in the UI, once in the DivePlannerModel. The reason
being that the user might switch units and we don't want to restrict
them to 1000 ft, when they switch from m. Time is simply restricted
in the UI, since we (thankfully) support only one time unit.

Fixes #2762.
commit 42569598 (2020-10-03): planner: limit depth and time of segments to sane values

Using non-sensical depth and times for segments in the planner may
lead to an unresponsive UI. Therefore limit depth to 1000 m/3000 ft
and time to 100 h. Limiting of depth is done twice: once to a
value of 3000 in the UI, once in the DivePlannerModel. The reason
being that the user might switch units and we don't want to restrict
them to 1000 ft, when they switch from m. Time is simply restricted
in the UI, since we (thankfully) support only one time unit.

Fixes #2762.
commit 154a5c62 (2020-10-03): planner: limit depth and time of segments to sane values

Using non-sensical depth and times for segments in the planner may
lead to an unresponsive UI. Therefore limit depth to 1000 m/3300 ft
and time to 100 h. Limiting of depth is done in settingsChanged()
since it has to adapt to the user changig their preferred units.

Fixes #2762.
PR #2997 (2020-10-03): planner: limit depth and time of segments to sane values
Using non-sensical depth and times for segments in the planner may
lead to an unresponsive UI. Therefore limit depth to 1000 m/3000 ft
and time to 600 min. Limiting of depth is done twice: once to a
value of 3000 in the UI, once in the DivePlannerModel. The reason
being that the user might switch units and we don't want to restrict
them to 1000 ft, when they switch from m. Time is simply restricted
in the UI, since we (thankfully) support only one time unit.

Fixes #2762.

### Describe the pull request:

- [ ] Bug fix
- [x] Functional ch [... cut]
closed 2020-10-03 by dirkhh