subsurface/subsurface #2776: Desktop: select visible dive locations takes very long

janmulder, 2020-04-20:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Given my logbook (1230 dives of which about 400 on 1 dive site) it can take 60 seconds to "select visible dive locations" on the map. At first, I thought that I found an endless loop but it simply takes unexpectedly long. 

Make sure the frequently visited dive site is on the map, and select the "select visible dive locations" from the map menu. Running with -v I see that for all matching dives a "Set the current dive site" line is printed. 

### Operating system:
OS: Arch Linux (up to date)
What device are you using? rather normal desktop machine

### Subsurface version:
What version of Subsurface are you running? current master
Does the same happen on another Subsurface version? don't no
Are you using official release, test build, or compiled yourself? self compiled
Provide Git hash if your are building Subsurface yourself. b0dfc0d0ffe7

### Steps to reproduce:
See above

### Current behavior:
Action takes very long.

### Expected behavior:
I would expect that selecting 400 dives out of a list of 1200 dives using the map method would take no more time than (multi-) selecting those same dives directly in the dive list.

atdotde, 2020-04-21:
My log file is much smaller, so I could not reproduce any significant delays. Maybe you could share your log with me (export and annonymize) by mail?
I played a bit with a profiler, though. It seems, the selection is changed one dive at a time which results in the main window tabs being recalculated quite often. Maybe this is something to look into.

atdotde, 2020-04-25:
Jan sent me his log file and I ran it through the profiler:
[image]
It seems, 

MapWidgetHelper::selectedDivesChanged

is called almost 60.000 times, each time updating all the information in the main window information tab.

Unfortunately, this is not directly called but through Qt's signalling model so I am lost at tracing the root of this, it might well be inside the QML part of the code. 

@neolit123 @bstoeger @dirkhh you seem to have written these parts of the code. Do you have any idea how to control how often this signal is emitted? Or alternatively, stop the action in the divest until the complete change of selection is done? It seems, the MapWidget already has such a mechanism in the isReady variable that prevents similar things happening at load time. Maybe we can set this to false during the update of the selection?

bstoeger, 2020-04-25:
First of all, I would apply #2782 - not because that should fix anything, but because it removes a different signal of the same name, so it makes debugging easier.

> MapWidgetHelper::selectedDivesChanged is called almost 60.000 times

That is strange, since `MapWidgetHelper::selectedDivesChanged` is only emitted at the end of the two functions. Thus, by reverse conclusion one of the two functions has to be called numerous times, no?

Then, I would suggest that you place two `printf`s before the emits and see which one (if not both) of these fire. Then I suggest putting a breakpoint there and look at the backtrace of say the first, the second and the 1000th invocation. That should give us an idea where the recursion happens.

> Or alternatively, stop the action in the divest until the complete change of selection is done? It seems, the MapWidget already has such a mechanism in the isReady variable that prevents similar things happening at load time. Maybe we can set this to false during the update of the selection?

If this can be avoided - please no. We have these kind of "ignore signal" states at a few places and invariably they turn out to be maintenance nightmares. Sometimes we can't avoid them, because the Qt API is intractable. However, if we can find/fix the root cause I would very much prefer that.

atdotde, 2020-04-25:
Some printf statements later, and stack tracing in the debugger, I have found the potential culprit: In divelistview.cpp, l.323 in function

void DiveListView::selectDive(QModelIndex idx, bool scrollto)

(which is called once per selected dive), there is a call to slectionChangeDone() (which is supposed to updated the main window. At least here, this is clearly too early. Commenting out that line makes at least makes the original "Select visible dives" fast and selects the correct dives. But I don't know if it is needed in other cases (Linus originally wrote that). Will post this to the mailing list.

bstoeger, 2020-04-25:
From a quick look, I think it can be removed:

First of all, I suggest making both `selectDive` functions private. Thus you can guarantee that they are not called from the outside.

Then you will see that there are two callers:

- One in `DiveListView::selectDives()`, however, that calls `selectedDivesChanges()` at the end - so it should be fine.
- The second caller is in `DiveListView::contextMenuEvent()`: This one will have to be audited. I'm not sure whether this is even relevant at all? Can it be removed? Should the selectedDivesChanged() be called here, or should it call `selectedDives()`?

If you prefer, I can look into these things.

bstoeger, 2020-04-25:
The second caller is especially peculiar:
```
                selectDive(get_divenr(current_dive), true);
                scrollTo(selectedIndexes().first());
```
It appears to be the only caller that calls `selectDive` with `scrollTo == true` and then proceeds to scroll somewhere else. That makes me believe that the whole `scrollTo` thing can be removed, no?

bstoeger, 2020-04-25:
Weirdly, removing that call, while probably correct, does not fix the extreme slowness of "select visible dive sites" for me.

bstoeger, 2020-04-25:
This here:
```
        QAbstractItemModel *m = model();
        QModelIndexList match = m->match(m->index(0, 0), DiveTripModelBase::DIVE_IDX, i, 2, Qt::MatchRecursive);
```
is also impossibly slow for me. I have an idea how to fix this, just not today anymore..

neolit123, 2020-04-25:
> is called almost 60.000 times, each time updating all the information in the main window information tab.

i remember benchmarking the map widget with thousands of markers when i wrote it, so this must have regressed in performance later.

> First of all, I would apply #2782 - not because that should fix anything, but because it removes a different signal of the same name, so it makes debugging easier.

+1 on merging that patch too.

i will review this in a bit:
https://github.com/Subsurface-divelog/subsurface/pull/2784

bstoeger, 2020-04-26:
With the changes in #2785, most of the time is now spent in this loop:
```
        QItemSelectionModel *s = selectionModel();
        for (const QModelIndex &index: indices) {
                s->select(index, QItemSelectionModel::Rows | QItemSelectionModel::Select);
                [...]
        }
```
This selects each item one-by-one, which causes numerous calls into the tree view. I did not find a way to select multiple items at once. Hopefully I am missing something? @neolit123, @thiagomacieira?

If this should really be the case, I see two possibilities:
1) Reimplement the QItemSelectionModel.
2) Intercept the calls to the QTreeView and call the QTreeView ourselves with the full list.

None of those appear all that appealing to me. Though the second one might be rather easy, since we already intercept the selectionChanged() calls. Still, if it is not necessary...

bstoeger, 2020-04-26:
Nevermind - I think I found it:
https://doc.qt.io/qt-5/qitemselectionmodel.html#select-1
`void QItemSelectionModel::select(const QItemSelection &selection, QItemSelectionModel::SelectionFlags command)`
Now I just have to find out how we can turn a QModelIndexList into a QItemSelection...

closed 2020-04-27 by janmulder