subsurface/subsurface #2788: Bogus air cylinders in planner

atdotde, 2020-04-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Start from an empty log, enter the planner. Though not in the cylinder table there are a number of air cylinders. You can see them by clicking on the "used gas" column in the dive planner points table in the selector. Or by adding a waypoint and right clicking on the handle in the profile window. The planner thinks those cylinder are real and tries to switch to air in dives that should not have air cylinders.

### Operating system:

### Subsurface version:

Current master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@bstoeger I guess this is a fallout of the new cylinder handling. To you see immediately where this could come from? Otherwise I will in investigate this further.

bstoeger, 2020-04-27:
I will check that out later today.

bstoeger, 2020-04-27:
That said, I couldn't reproduce that easily. Tried:
- Started without log. Went to planner - only saw one air cylinder.
- Started with a test log. Went into planner - got the same gases as the last dive.
- Started with a test log. Closed log. Went into planner - only saw one air cylinder.

In all three cases The plannerpoints and the profile context menu were consistent with the cylinder list.

atdotde, 2020-04-27:
Just realized: You have to have the surface segment >0. That used to work having air there as a special gas.

bstoeger, 2020-04-27:
Thanks - I can reproduce this now, will try to debug later.

bstoeger, 2020-04-27:
The problem was introduced in 7c9f46acd202121e67557bb634961ef17a9f6c1f. A cylinder is added every time the surface interval is changed. This is obviously wrong. However, the fundamental problem is that the old code was pointing to a non-existing cylinder to simulate breathing non-compressed air(?). This cylinder was not shown to the user. Since removing the MAX_CYLINDERS restriction, we obviously don't support this anymore. I will have to think about this. Should we add a cylinder and mark it as "don't show"? Or somehow use a special index meaning "uncompressed air"?

atdotde, 2020-04-27:
Yes, I thought we had settled on something like this. I would call it "surface air".

atdotde, 2020-04-27:
Looking at the code, we add a cylinder with cylinder_use = NOT_USED. You think, it's this cylinder?

bstoeger, 2020-04-27:
Yes. My quick fix was to check if we have an empty air cylinder and only add a new cylinder if that doesn't exist.

However, that still had two problems: the cylinder-model didn't realize that there was a new cylinder. And even if we fix that, the user then sees an ugly empty cylinder in the list.

bstoeger, 2020-04-27:
My intuition would be to add a planner segment with cylinder index "-1", which has the meaning of "no cylinder". However, I fear the ensuing monster-breakage. This will have to be audited very carefully and I am not familiar with the planner at all (neither the code, nor the physics behind it).

atdotde, 2020-04-27:
I would assume, handling a negative index would be a nightmare. What's wrong with this special tag in the cylinder_use field? My assumption was, that the planner always starts with an empty cylinder list, then adds those from the cylinder model and finally, after the plan is complete this last surface cylinder. So they should not multiply.
Right now, I have to do some real(™) work. Will look into this later today.

bstoeger, 2020-04-27:
There are a few problems with adding a dummy-cylinder:

If you add it in the planner, the model does not know of it and you end up with an inconsistent state. The old code did not have that problem, because the empty-cylinder was recognized by the model as "end of list", so effectively no inconsistent state was produced. You probably can get away with such an inconsistent state if you are very careful, but it doesn't sound like a good idea. Notably, every time a cylinder is added, the surface segment has to be rebuilt, because it will now point to an overwritten cylinder.

So to avoid such difficulties you have to tell the model that you added a cylinder form the planner, which means calling from C into C++, which is always painful. Then, if you don't want to bother the user with these internal matters, you have to filter out the cylinder from the view. Which means adding another CylinderFilterModel. I remember adding this was not fun at all.

> I would assume, handling a negative index would be a nightmare

I'm not intimate with the code, so it's hard to tell for me. I have a feeling that with some auditing it should be doable - if it weren't so close to release. However, in the long-run I think it is the correct way to go: No cylinder means no cylinder, not some strange fake cylinder. Think about free-dives. Though I reckon the planner isn't used for these...?

bstoeger, 2020-04-27:
Another reason why I'm not a fan of the dummy-cylinder is that this probably bleeds into other parts of the code. You now have to handle that special cylinder in the equipment tab. And perhaps on the profile or the statistics. An index of "-1" for the segment would (hopefully) stay in the planner.

bstoeger, 2020-04-27:
After a (very short) audit it appears to me that the only problematic part of the code with respect to a negative `cylinder_id` is `addStop`, which already interprets `cylinder_id == -1` as use same cylinder. The rest seems rather manageable.

I still feel that will be easier than adding a fake cylinder, because that will come out in numerous places. Of course, you know that part of the code much better than me.

atdotde, 2020-04-27:
I guess, something else is going wrong. Have a look at what https://github.com/atdotde/subsurface/tree/debugbogusgas prints when you do the following:

Start from empty log, enter the planner, make sure surface segment is non null
make the first gas trimix 18/45, add a second gas (this is added as air).
Make that second gas EAN50 and look at the output:

As you can see, the EAN50 is marked not used while there is a used air cylinder. This is clearly wrong.

atdotde, 2020-04-27:
I should add, to what I wrote in the commit message: The planner then added segments to the dive plan and in the end a dive (displayed dive) is created from the dive plan. It seems to me, this new dive somehow does not start with an empty list of cylinders (to which those form the plan are added).

bstoeger, 2020-04-27:
I'm currently in the vineyards, will check it out later. I have a very disgusting idea, how temporarily fix the air problem.

dirkhh, 2020-04-27:
> I'm currently in the vineyards, will check it out later. I have a very disgusting idea, how temporarily fix the air problem.

Depending on what "in the vineyards" means, maybe it's good if you sleep on that idea? 
:-) :-) :-) :-)

I hope you are socially distancing in those vineyards.

bstoeger, 2020-04-27:
I have to disappoint you - only cycling in the hills by myself. Perhaps surprisingly, we have quite a sizable wine production within city limits (http://www.wienerwein.at/) and it's one of the few nice places in town. Apparently, restaurants may reopen mid-may and I suppose that includes the "Heurige". Then I might consume their produce in this nice setting.

Now back to topic... ;)

bstoeger, 2020-04-27:
> I guess, something else is going wrong.

This does not surprise me at all: By adding a surface interval, the planner manipulates the cylinder table without the model knowing it. Hence, the state becomes inconsistent and when you edit a cylinder it will trample over the fake-cylinder and chaos ensues. At least this is my theory. First, we have to fix the air-thing, then we can look at that particular failure mode.

atdotde, 2020-04-27:
Hmm. Let's look for example as a reference at the state at 0e55739f03d15b93a678e29bb

There was a function show_cylinder() that determined if a cylinder in the dive should show up in the cylinder model. The surface segment was using a cylinder that was not to be shown in this sense. This possibility was eliminated with the new cylinder handling. Would it be possible to get this back, unshown cylinders? Then the planner could use it for the surface segment. It would only have to be created when it's not there, yet. I still think, one could use the cylinder_use field to signal such a special cylinder.

bstoeger, 2020-04-27:
> There was a function show_cylinder() that determined if a cylinder in the dive should show up in the cylinder model.

That local function was totally broken and never really worked. If it returned true for anything but the last cylinder, the model would get terminally confused. We do have a `CylindersModelFiltered` that does this correctly now. You could clone that and use it to hide whatever you want.

The real reason that the old code worked was because the fake air cylinder was basically non-existent for the rest of the code in the sense that it returned true for `cylinder_nodata(cyl)`. This was the marker for "doesn't exist". See for example the code in `getGasList` in `qt-models/models.cpp`.

As I said above: if you want to go this route, you have to make sure that the models are updated when you add a new cylinder in the planner. The old code did not have this data inconsistency problem, because the fake cylinder simply didn't exist, so the data was not changed in that sense. Effectively, you simply created an index past the last cylinder, which is what #2794 tries to emulate. In contrast, now the `cylinder_table->nr` counter is increased when you add a cylinder, which makes core and model inconsistent.

I still believe that adding the ability to switch to "no gas" will ultimately be the easier route. We could even add a special icon for that. However, in the end, it's really up to you

bstoeger, 2020-05-03:
I think we can close (even if the current resolution is not ideal)...?

closed 2020-05-03 by atdotde