subsurface/subsurface #2808: Dive date is always a day after the dive

mikeebowen, 2020-05-03:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Using a Cressi Leonardo, the date is always one day after the actual dive occurs. For example, if the dive was on May 5, 2020 the app will show the dive on May 6, 2020. The time is correct.
### Operating system:

OS: macOS Catalina 10.15.4

MacBookPro11,3

### Subsurface version:

4.9.3

Unknown

official release

### Steps to reproduce:

1) Download a dive 
2) View that the date is a day after the dive actually occurred.
### Current behavior:

The date is a day after the actual date
### Expected behavior:

The date will be correct.
### Additional information:

### Mentions:

dirkhh, 2020-05-03:
Latest version (as of yesterday :grin: ) is 4.9.4 - but I don't think that would necessarily change your result... it just makes testing for us easier if you are on the latest version.

Can you please try again, and in the Download dialog select "Save libdivecomputer logfile", then download a dive and send us logfile as well as the dive that was downloaded (you can export individual dives from the File-Export menu)

mikeebowen, 2020-05-04:
I updated to 4.9.4 ( I like the new information view 😄 ) but the date issue is still there. Where can I send you the logfile? Github won't let me attach it to the issue.

Subsurface-divelog, 2020-05-04:
Just add it to a zip file and you should be able to upload. Or email to dirk@hohndel.org

mikeebowen, 2020-05-05:
[dive-log.zip](https://github.com/subsurface/subsurface/files/4578426/dive-log.zip)
Thanks for the tip! I didn't know about the zip files. Here is the logfile and a dive from 4/25 that shows in the app as 4/26

bstoeger, 2020-05-05:
> Here is the logfile and a dive from 4/25 that shows in the app as 4/26

The log file clearly says:
```
<dive number='23' date='2020-04-26' time='13:08:00' duration='33:40 min'>
```
Hence this appears to be an import, not a display problem.

You can correct the date of multiple dives in the following way: Select all dives you want to correct and then select the correct date for the displayed dive. All other dives will be adjusted with the same offset. Take care to check the results before saving, though!

dirkhh, 2020-05-06:
@jefdriesen any idea what might be causing this?

mikeebowen, 2020-05-06:
Thanks for the help. You're right that clearly has the wrong date. I'll correct the dates manually. The problem must be with the Cressi Leonardo and not the app. Thanks for creating this awesome app, it's sooooo much better than the Cressi app!

jefdriesen, 2020-05-06:
The libdivecomputer output is:
```
2020-05-03 21:01:00
2020-05-03 19:27:00
2020-04-26 13:08:00
2020-04-26 11:31:00
2020-04-19 12:46:00
2020-04-19 10:55:00
2020-04-19 10:46:00
2020-04-19 10:41:00
2020-04-19 10:36:00
2020-04-06 15:39:00
2020-04-06 13:35:00
2020-04-05 12:47:00
2020-04-05 11:18:00
2020-03-29 13:06:00
2020-03-29 11:01:00
2020-03-23 11:44:00
2020-03-23 10:02:00
2020-03-22 08:48:00
2020-02-17 11:48:00
2020-01-19 17:04:00
2020-01-19 16:03:00
2020-01-19 15:55:00
2020-01-19 15:35:00
2020-01-19 15:26:00
2019-11-03 22:19:00
```
It's a direct translation of the raw data, so there is not much that can go wrong. What does the Cressi application show?

mikeebowen, 2020-05-06:
The Cressi application doesn't work at all. It doesn't display any data. It is completely broken. The dive watch itself shows the correct dates and times when you scroll through the logs.

mikeebowen, 2020-05-08:
I'm closing this, since it is an issue with my dive computer and not with the app. Thanks again for all your help!

closed 2020-05-08 by mikeebowen