subsurface/subsurface #2829: Add SAC to the export dive details as csv

p-neves, 2020-05-12:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X ] New feature request
- [ ] Discussion request

### Issue long description:

It would be nice to have the "SAC" field on the list of fields exported on the csv dive details file

### Operating system:

### Subsurface version:

4.9.4

Self compiled

80-gce7cc7970540

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mturkia, 2020-05-13:
This will require some more effort as the export to CSV is based on saved XML file, and SAC is not stored in it (as Subsurface calculates that on-the-fly). Would it make sense to store SAC on the XML but not parse it, when opening a Subsurface XML file? This would allow XSLT to export SAC.

dirkhh, 2020-05-13:
Interesting. I had wondered about that since SAC is indeed purely a calculated value.
Hmm. Can you poke Linus on this one? He doesn't respond to @torvalds mentions (gets way too many), so just send him an email with a link to this github issue...

torvalds, 2020-05-14:
https://github.com/subsurface/subsurface/pull/2835

Close this if that works for you.

PR #2846 (2020-05-18): Csv sac

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [x] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Adds SAC to be included on dive detail CSV export.

### Changes made:

### Related issues:

Fixes #2829

### Additional information:

### Release note:

### Documentation change:
commit 0525f315 (2020-05-18): CSV export: include SAC when exporting dive details

Simply adds SAC field to CSV export of dive details.

Fixes #2829
closed 2020-05-18 by dirkhh