subsurface/subsurface #2830: Issues with CSV summary dive details - multitank dives

p-neves, 2020-05-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When exporting summary dive details as csv, information on multiple tanks (if used) is not exported. Specifically, information regarding cylinders used (cylinder size, startpressure, endpressure, o2 and he) is only shown for the first cylinder on the equipment tab. 

### Operating system:

Arch Linux
en_US.UTF-8

### Subsurface version:

9.4.4

Compiled myself

gce-7cc7970540

### Steps to reproduce:

1) Select dives (in this case I've selected dives with more than one tank)
2) Select File/Export and then "CSV summary dive details". Check the selection button "Selected dives"
3) Open the exported csv file and verify the missing info

### Current behavior:

The file only shows information (cylinder size, startpressure, endpressure, o2 and he) for the first cylinder on the equipment tab

### Expected behavior:

Information on all used cylinders should be exported (including -- feature request -- "SAC" for each cylinder.

### Additional information:

### Mentions:

dirkhh, 2020-05-12:
@mturkia that sounds like it would fall into your wheelhouse (as does #2829)

mturkia, 2020-05-13:
Jup, I'll try to take a look. The initial implementation was indeed for the first cylinder only as that is the amount of cylinders I use.

closed 2020-05-26 by dirkhh

p-neves, 2020-05-26:
When I export my data I get hundreds of cylinder columns, which then exceeds the column limit for e.g. libreoffice. Can't we limit the number of tanks to a reasonable number (say 10?).

thanks

mturkia, 2020-05-27:
Can you send me a sample log in subsurface xml? That would demonstrate the issue? I think there is a bug somewhere.

commit 889f36a5 (2020-05-26): Multi cylinder support for CSV export

Fixes #2830