subsurface/subsurface #2834: Weird parenting-like problem in QProgressDialog

janmulder, 2020-05-14:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I save my logbook locally (which is a local git store), and I save my logbook after that to the cloud storage, the progress dialog shows up in the upper-left corner of my screen. As if its not parented to the mainwindow. This it just a visual glitch, but it annoys me. 

### Operating system:
What OS are you running, including OS version and the language that you are running in: Arch Linux
What device are you using?: generic desktop machine

### Subsurface version:
What version of Subsurface are you running?: master
Does the same happen on another Subsurface version?: Not sure, looks like something Qt related and I have seen this behavior already for a long time.
Are you using official release, test build, or compiled yourself? self compiled
Provide Git hash if your are building Subsurface yourself: nor relevant

### Steps to reproduce:
Open Subsurface, edit a dive, save locally. Now, save to cloud, en see that the progress dialog appears in the wrong place

### Current behavior:
see above

### Expected behavior:
obvious

### Additional information:
I think I have a trivial, 1 line fix for this. but at this point I'm interested if anyone can reproduce this behavior.

dirkhh, 2020-12-12:
Oh looky, seven months went by.
I remember looking at this when you posted it and trying to reproduce it.
I still can't seem to make this happen, but of course by now we are on a different Qt version in Arch and a thousand other things have changed.
Do you still see this?

janmulder, 2020-12-13:
I never had a deterministic way to provoke this behavior, and that does not help. But I just played around in this area and I can't reproduce it anymore. So I propose to close this, and I can always re-open when I see it again.

closed 2020-12-13 by janmulder