subsurface/subsurface #2838: I can't get the 4.9.4 to open on a mac running 10.11.16

gr8tfl, 2020-05-15:
I can't get the 4.9.4 to open on a mac running 10.11.16

### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-05-15:
macOS 10.11 is no longer supported in the official binaries since the Qt libraries that were used to build the binaries require at least macOS 10.12.
It should still be possible to build on macOS 10.11 using Qt 5.11 - but that's not something I currently have the bandwidth to do.

closed 2020-05-25 by dirkhh

dirkhh, 2022-03-30:
Hi @gr8tfl -- and thanks for the sponsorship.
I have no way to get an email address for people who send money... often I can figure out who it is from their GitHub profile, but with you that failed. So an awkward Thank You here in the only issue you ever opened :)

gr8tfl, 2022-03-30:
If figuring out is important I unmask myself. 

Thanks for Subsurface. If I could vote on how the very small contribution is used it would be on good coffee in the morning or beer in the afternoon, but I give for you to use however you like. 

Thanks again

Max Erenberg
+1 773 771 3750
***@***.***
Sent from a mobile device’s