subsurface/subsurface #2842: filter bug in Subsurface-mobile when filtering for tags

dirkhh, 2020-05-16:
### Describe the issue:

- [x] Bug

### Issue long description:

From Hartley H: if I filter on a tag, such as “wreck”, it only shows dives where the ONLY tag is wreck.  Most of my dives have comma separated tags and wreck dives will have a tag of “boat, wreck” as an example.  Those dive are not found in the filter list. 

### Operating system:

iOS 12.4

### Subsurface version:

3.0.6(4.9.4.93) - but same behavior in earlier versions

### Mentions:

@bstoeger (don't know Hartley's GitHub user name)

h2sk1, 2020-05-16:
Thanks Dirk.  To make things confusing to all, my GitHub isn't related to my subsurface handle.   Hartley

closed 2020-05-16 by dirkhh

h2sk1, 2020-05-18:
New iOS beta works for me.  Thanks.

commit c184f3e9 (2020-05-16): filter: properly search for tags

The old code used get_taglist_string() and split the resulting
string at commas to get the list of tags. This was wrong for two
reasons:
1) It was buggy. Every tag but the first would start with a leading
   space and thus not be found.
2) It was inefficient. The tag list was concatenated, just to be split
   again.

Turn the tag list directly into a QStringList and remove whitespace
for good measure.

Fixes #2842.

Reported-by: Hartley Horwitz <hhrwtz@gmail.com>
commit 1944ace1 (2020-05-16): filter: properly search for tags

The old code used get_taglist_string() and split the resulting
string at commas to get the list of tags. This was wrong for two
reasons:
1) It was buggy. Every tag but the first would start with a leading
   space and thus not be found.
2) It was inefficient. The tag list was concatenated, just to be split
   again.

Turn the tag list directly into a QStringList and remove whitespace
for good measure.

Fixes #2842.

Reported-by: Hartley Horwitz <hhrwtz@gmail.com>
PR #2843 (2020-05-16): Fix tags filter

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

This fixes #2842. There were actually two bugs, see commit description.

### Changes made:

1) Index tags in fulltext filter.
2) construct tag list directly - don't split a concatenated string.
### Related issues:

Fixes #2842.
### Mentions:

@dirkhh @h2sk1
commit a7440ce2 (2020-05-16): filter: properly search for tags

The old code used get_taglist_string() and split the resulting
string at commas to get the list of tags. This was wrong for two
reasons:
1) It was buggy. Every tag but the first would start with a leading
   space and thus not be found.
2) It was inefficient. The tag list was concatenated, just to be split
   again.

Turn the tag list directly into a QStringList and remove whitespace
for good measure.

Fixes #2842.

Reported-by: Hartley Horwitz <hhrwtz@gmail.com>