subsurface/subsurface #2845: 4.9.4 Update not opening on 10.15.4 macOS Catalina

gkeeler7, 2020-05-17:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Window opens that says “Subsurface” can’t be opened because Apple cannot check it for malicious software. This software needs to be updated. Contact the developer for more information."

### Operating system:

10.15.4 macOS Catalina

Macbook Pro

### Subsurface version:
 4.9.4
 Opened pervious version fine
 no

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-05-17:
Yeah, the notarization service currently doesn't want to talk to me and Apple support is its usual helpful self if you aren't completely integrated into their walled garden.
You should be able to go to your Global Preferences / Security and white list the app (after you started it and got the failure you describe). I tried this on a clean Catalina install in a VM and it worked fine.

gkeeler7, 2020-05-18:
thanks, Dirk, that was a successful workaround. Cheers!

closed 2020-05-18 by gkeeler7