subsurface/subsurface #2858: Unable to connect to cloud on new Mac installation

henrik242, 2020-05-25:
I installed Subsurface 4.9.4 from Homebrew on macOS 10.15.4.

The File -> Open/Save cloud menu entries are disabled.  Adding my credentials to the preferences doesn't seem to do anything.  The web version at https://cloud.subsurface-divelog.org/user/henrik@XXX.no/dives.html works fine.

Here's the application output:
```
XXX@work:~🐤 /Applications/Subsurface.app/Contents/MacOS/Subsurface  -vvv
Subsurface v4.9.4,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (7882ba423cd87569f9c9361fa999917e9ced1083)
built with Qt Version 5.13.2, runtime from Qt Version 5.13.2
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 2.1."
QFont::setPointSizeF: Point size <= 0 (0.000000), must be greater than 0
QFont::setPointSizeF: Point size <= 0 (0.000000), must be greater than 0
QFont::setPointSizeF: Point size <= 0 (0.000000), must be greater than 0
can't find Qt localization for locale "nb-NO" searching in "/Applications/Subsurface.app/Contents/translations"
loading dive data from ("/Users/XXX/Library/Application Support/Subsurface/XXX.xml")

File locations:

cloud URL set as "https://cloud.subsurface-divelog.org//git/henrik@XXX.no[henrik@XXX.no]"
Local git storage: /Users/XXX/Library/Application Support/Subsurface/cloudstorage/360cb7089fc8f3fe
Cloud URL: https://cloud.subsurface-divelog.org//git/henrik@XXX.no[henrik@XXX.no]
Image filename table: /Users/XXX/Library/Application Support/Subsurface/hashes
Local picture directory: /Users/XXX/Library/Application Support/Subsurface/picturedata/

Received error response from cloud storage backend: "Error transferring https://cloud.subsurface-divelog.org//storage - server replied: Service Unavailable"
Completed connection with cloud storage backend, response "<!DOCTYPE HTML PUBLIC \"-//IETF//DTD HTML 2.0//EN\">\n<html><head>\n<title>503 Service Unavailable</title>\n</head><body>\n<h1>Service Unavailable</h1>\n<p>The server is temporarily unable to service your\nrequest due to maintenance downtime or capacity\nproblems. Please try again later.</p>\n<hr>\n<address>Apache/2.4.29 (Ubuntu) Server at cloud.subsurface-divelog.org Port 443</address>\n</body></html>\n"
Received error response from cloud storage backend: "Error transferring https://cloud.subsurface-divelog.org//storage - server replied: Service Unavailable"
Completed connection with cloud storage backend, response "<!DOCTYPE HTML PUBLIC \"-//IETF//DTD HTML 2.0//EN\">\n<html><head>\n<title>503 Service Unavailable</title>\n</head><body>\n<h1>Service Unavailable</h1>\n<p>The server is temporarily unable to service your\nrequest due to maintenance downtime or capacity\nproblems. Please try again later.</p>\n<hr>\n<address>Apache/2.4.29 (Ubuntu) Server at cloud.subsurface-divelog.org Port 443</address>\n</body></html>\n"
```

Wardy-uk, 2020-05-25:
this problem seems to exist across all devices - I am getting the same issue with the IOS app, Android app, as well as Mac

dirkhh, 2020-05-25:
This should be fixed now.

Longer answer. Server issues yesterday. Ended up migrating to the hot standby. Had problems with that. Things seemed to work just before I went to bed, only to fall flat on their face a few minutes later. The problem was the daemon that Subsurface interacts with in order to verify git accesses.

closed 2020-05-25 by dirkhh

henrik242, 2020-05-26:
> This should be fixed now.

Great work!