subsurface/subsurface #2863: Same Problem with Version 3.0.6.

sausimausi, 2020-05-28:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

sausimausi, 2020-05-28:
I have the same problem: on android mobile apps no dives show with android version 3.0.6

dirkhh, 2020-05-28:
That's not really a useful issue report.
It doesn't tell us what's not working, it doesn't include a log.
What do you expect the developers to do with this, other than waste their limited time on responding and asking you clarifying questions?

closed 2020-06-07 by dirkhh