subsurface/subsurface #2866: gasmix -1 error downloading dive from Shearwater Perdix AI

johnsgill3, 2020-05-29:
### Describe the issue:
`gasmix - 1 for tank 0 doesn't match`
- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Every time that I attempt to download dives from Shearwater Perdix AI over Bluetooth I am getting the following error banner that appears: 
[image]. I am able to still successfully download the dive and import it, but the error is confusing and annoying.

### Operating system:

Mac OS X 10.15.4, English

Shearwater Perdix AI FWv72

### Subsurface version:

4.9.4

Haven't tried any other versions

Release Build

### Steps to reproduce:

1) Click `Import`
1) Click `Import from Dive Computer`
1) Select `Choose Bluetooth download mode`
1) Click `...`
1) Select `Perdix {<>}`
1) Click `Save`
1) Click `Download`
1) Error banner shows up in main window

### Current behavior:

Error banner appears importing dives from computer

### Expected behavior:

No error banner

### Additional information:

### Mentions:

hhmeyer, 2020-06-09:
Just chiming in to say I have the same issue.

bstoeger, 2020-06-12:
Hm. The obvious suspect would be the MAX_CYLINDERS removal: 7c9f46acd2

However, I don't see how the changes would affect the local `tank` structure in this file.

If you say that the gas-information is imported properly, I would cautiously simply remove the warning. That said, I'm not familiar at all with this part of the code.

@dirkhh: You wrote the warning (4be7604634), so I reckon that you would understand best what this is about.

bstoeger, 2020-06-13:
I don't have the hardware, so I can't test it, but hopefully #2874 fixes the problem.

PR #2874 (2020-06-13): download: don't warn on unknown gasmixes
Apparently libdivecomputer can return DC_GASMIX_UNKNOWN when
fetching tank info with
  dc_parser_get_field(parser, DC_FIELD_TANK, i, &tank);
This caused emission of a warning, which was annoying users.
Disable the warning in that case.

Fixes #2866

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Don't show warning when we get `DC_GASMIX_UNKNOWN` from libdivecomputer.

### Related [... cut]
commit 23e88391 (2020-06-13): download: don't warn on unknown gasmixes

Apparently libdivecomputer can return DC_GASMIX_UNKNOWN when
fetching tank info with
  dc_parser_get_field(parser, DC_FIELD_TANK, i, &tank);
This caused emission of a warning, which was annoying users.
Disable the warning in that case.

Fixes #2866
closed 2020-06-13 by dirkhh