subsurface/subsurface #2869: Cannot sync with cloud server, working with offline copy

Stefan-DH, 2020-06-01:
### Describe the issue: 
Cannot sync with cloud server, working with offline copy

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
when trying to open the cloud storage, I'm getting the error "Cannot sync with cloud server, working with offline copy" and no data is downloaded.
In the configuration I can enter the required credentials (and they are accepted, because when I enter a faulty password, I get the message that the password is fault. if I enter the right credentials, it is OK)
if I click the option "open cloud storage" I see the following steps:
- sync with cloud
- wait to connect
- successful connected
than the popup closes and I get the mentioned error at the bottom of the screen/application

### Operating system:
using Windows 10, tried it on 2 different machines.
I have also the subsurface mobile running on Android, and that one is working perfectly.
maybe worth mentioning: till last week I had another windows 10 computer which I used to work with. via that computer I entered all my dives without a problem and could sync with the cloud. now I switched PC's (and I don't have the old one available) installed the subsurface software on it but I'm not able to sync with the cloud

### Subsurface version:
subsurface on windows: 4.9.4
mobile version on android: version 3.0.7(4.9.4.144)

### Steps to reproduce:
1. open software 
2. file -> open cloud storage

### Current behavior:
Cannot sync with cloud server, working with offline copy
(no offline (thus local) data available)

### Expected behavior:
want to download the already registered dives from the cloud platform to continue with entering new dives

### Additional information:

### Mentions:

fabo137, 2020-06-01:
Same problem over here. 

Was entering my dives today. Entered one dive after each other in online mode. By hitting CMD+S it was pushing the changes to cloud. Worked fine. At one point it stopped working and showing the error message: "Failure taking cloud storage online".

Now works with offline copy ( local cache)

What is also maybe worth to mention is that the successfully synced dives were not fetcht from the mobile. The mobile still has the state from yesterday even when performing a manual syncp
The mobile doesn't show any error during the sync process. 

I've yesteready reinstalled both, mobile and Desktop app

I'm using macOS catalina 10.15.4

Will now try to uninstall the application on both devices, reset the password again and reinstall.

fabo137, 2020-06-01:
Update:

after uninstalling App on Android, reinstall, resetting password and login, no dives at all in the list.

Reinstalled the app again, verified all app data was purged, same behaviour:

Device information: 
Model numbger: SM-G975F

fabo137, 2020-06-01:
worked again 10 minutes before stopped working 16:42.

Now im completely confused :S

dirkhh, 2020-06-01:
The only person who can debug this happens to be me - since I run the cloud server.
We had a fatal crash about a ten days ago and I migrated everything to the hot standby server.
It seems to work fine for most people, but a small number of people appear to have problems.

Please send me (dirk@hohndel.org) email FROM the account that you use for cloud storage and let me help you. If you are using Subsurface-mobile, what works great is to use Help->Ask for support (as that email also goes just to me) as that includes important logs. But if the email client on your phone doesn't use the same From address, I'll still need an email from that account as well

dirkhh, 2020-06-07:
I believe all of these cases are now fixed. If you still have issues, please contact me in email so I can chase those down. Thanks

closed 2020-06-07 by dirkhh