subsurface/subsurface #2877: Multiple tank, pressure issue

BartJV, 2020-06-14:
(I'm new to reporting an issue, so I hope this is ok):

I use two tanks with a pressure sensor on each.  It seems the second tank's data is not available due to the following bug:

The following line resets the pi->nr value after loading a profile into memory
https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.c#L645

but as a result, breaks array access in the following set/get functions. This breaks future access to pressure data for any additional tanks.

https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.h#L110-L113

I've replace line 645 with the following, as a temporary workaround, but its just a hack, so I didn't want to submit it.

```
//Re-order the pressures...
    int nrActual = pi->nr;
    pi->nr = idx;
    for( int cyl = 0; cyl < pi->nr_cylinders; cyl++){
        for(int sample = 0; sample < pi->nr; sample++){
            for(int sensor = 0; sensor < NUM_PLOT_PRESSURES; sensor++){
                pi->pressures[cyl * pi->nr + sample].data[ sensor ] =
                        pi->pressures[cyl * nrActual + sample].data[sensor];

            }
        }
    }
```

mturkia, 2020-06-15:
The pi->nr value is not actually reset but increased by at least 2 as idx is increased in the previous lines (idx++).
```
	plot_data[idx++].sec = lasttime + 1;
	plot_data[idx++].sec = lasttime + 2;
	pi->nr = idx;
```

And this assignment is the last thing we do within the function, so all the data is already filled and we are exiting populate_plot_entries next.

I have not looked into this part of the code in a long long time, but the sample pressures are recorded earlier in the function (line 604):
```
		if (sample->pressure[0].mbar)
			set_plot_pressure_data(pi, idx, SENSOR_PR, sample->sensor[0], sample->pressure[0].mbar);
		if (sample->pressure[1].mbar)
			set_plot_pressure_data(pi, idx, SENSOR_PR, sample->sensor[1], sample->pressure[1].mbar);
```
This seems to be hardcoded for max of 2 sensors/cylinders, so could this be the problem? So maybe have a loop on line 604 to fill in the sensor/cylinder data for the amount we actually have?

BartJV, 2020-06-15:
pi->nr is actually decreased compared to its initial value. 
https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.c#L527

As a result, the second (and onwards) tank is no longer at the expected offset.
I agree that the data is already stored at this stage; the second tank starts at **cylinder * pi->nr**
https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.h#L112

After line 645 though, that offset is no longer valid and the second tank data is now expected at **cylinder * idx ** from line 645
This breaks get_plot_pressure_data.

mturkia, 2020-06-16:
> pi->nr is actually decreased compared to its initial value.
> 
> https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.c#L527

True, the initial value is for the worst case scenario, and the value stored at the end is the actual value of samples and events plus 1.

> As a result, the second (and onwards) tank is no longer at the expected offset.
> I agree that the data is stored prior to line 645. The Nth tank data starts at **cylinder * pi->nr**
> 
> https://github.com/subsurface/subsurface/blob/b01361170770a1cd263c57e856437e5f22c794df/core/profile.h#L112
> 
> After line 645 though, that offset is no longer valid and the second tank data is now expected at **cylinder * idx ** from line 645
> This breaks get_plot_pressure_data for all but the 0th tank.

Now that looking at the code, I am thinking the problem would be in profile.h. The get_plot_pressure_data and set_plot_pressure_data should use pi->nr_cylinders instead of pi->nr. Unfortunately I do not have test data that demonstrate the issue and prove me right or wrong.

BartJV, 2020-06-16:
Agreed; profile.h would work fine if pi->nr wasn't modified in profile.c.

The pressure data has a size of "pi->nr" in memory, lets say 100 pressures.
Tank0 has pressures from 0-99, starting at **0** * pi->nr
Tank1 has pressures from 100-199, starting at **1** * pi->nr

After pi->nr is decreased to let's say 75, we end up with
Tank0 has pressures from 0-74, starting at **0** * pi->nr(75)
Tank1 has pressures from 75-149, starting at **1** * pi->nr(75)

My solution would tend to be having an array for each cylinder. I'll try to add a dataset.

bstoeger, 2020-06-29:
Could you please test this quick-fix:
```
diff --git a/core/profile.c b/core/profile.c
index 1cc73f158..ded275ca5 100644
--- a/core/profile.c
+++ b/core/profile.c
@@ -512,6 +512,7 @@ static void populate_plot_entries(struct dive *dive, struct divecomputer *dc, st
 	int idx, maxtime, nr, i;
 	int lastdepth, lasttime, lasttemp = 0;
 	struct plot_data *plot_data;
+	struct plot_pressure_data *final_pressure_data;
 	struct event *ev = dc->events;
 	maxtime = pi->maxtime;
 
@@ -642,7 +643,20 @@ static void populate_plot_entries(struct dive *dive, struct divecomputer *dc, st
 	/* Add two final surface events */
 	plot_data[idx++].sec = lasttime + 1;
 	plot_data[idx++].sec = lasttime + 2;
+
+	/* We probably used less plot_data entries than needed.
+	 * "pi->nr" is the number of the allocated entries and "idx" is the number
+	 * of needed entries. Now, let's alloc an array of the actually needed size
+	 * and continue using that.
+	 */
+	final_pressure_data = calloc(idx * (size_t)pi->nr_cylinders, sizeof(struct plot_pressure_data));
+	for (int i = 0; i < pi->nr_cylinders; ++i) {
+		for (int j = 0; j < idx; ++j)
+			final_pressure_data[i * idx + j] = pi->pressures[i * pi->nr + j];
+	}
+	free(pi->pressures);
 	pi->nr = idx;
+	pi->pressures = final_pressure_data;
 }
 
 /*
```
Thank you.

bstoeger, 2020-06-29:
Also, I don't have your e-mail address for bug-reporting credits.

closed 2020-07-02 by dirkhh