subsurface/subsurface #2882: Settings does not save "Dive Log" setting when set to cloud

jebbgrenham, 2020-06-21:
### Describe the issue:

- [X ] Bug

### Issue long description:
0. Already have cloud credentials in Subsurface and working using "Open Cloud Storage" in the file menu.
1. Open the preferences, click "Dive Log" and select "Cloud" as the default log file.
2. Click save.
3. Restart subsurface

Expected behavior: Dive log opens automatically from cloud
Actual behavior: Subsurface does not open cloud storage, instead settings have reverted to "Local Default File".

### Operating system:
Mint 19.3 English.

### Subsurface version:
4.9.6

### Steps to reproduce:
See above

### Current behavior:
See above

### Expected behavior:
See above

### Additional information:

### Mentions:

jebbgrenham, 2020-06-21:
I've just realized this bug applies to more preferences than just the cloud setting. Every preference I've tried to set so far is completely disregarded when subsurface is close and re-opened...

dirkhh, 2020-06-21:
In most cases when this happens the user has run Subsurface as `root` at some point and the preferences are owned by `root` and can't be overwritten. Could this be the case here?

jebbgrenham, 2020-06-25:
Just checked, it is not the case. All owned by my user.

jebbgrenham, 2020-06-25:
Oops, no it is the case. I was looking at .subsurface where I expected prefs to be; not .config/Subsurface. It was owned by root and I do recall that root is required to download from a suunto zoop (adding to dialout doesn't work for that particular computer). Anyway; chown to the rescue. All seems well now, thanks.

mturkia, 2020-06-25:
Check the permissions of the device used for downloading from Zoop. What is the group and file permissions? If the device is i.e. owned by root and group is root, and others don’t have permission to write, you need to set up udev rules to set proper permissions. This is probably documented in our user manual to some extent. And our faq has also some instructions that can be adapted to your device.

closed 2020-12-12 by dirkhh