subsurface/subsurface #2883: Dive merge corrupts log file.

BartJV, 2020-06-23:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Merging two dives with tank pressure data included corrupts the dive log file (in some cases) preventing it from being opened again.

### Operating system:
Windows and Ubuntu, Perdix AI

### Subsurface version:
4.9.6, Official release

### Steps to reproduce:

[testCase.txt](https://github.com/subsurface/subsurface/files/4822331/testCase.txt)

-Open attached log file.
-Merge dive 471 and 470
-4 tanks are created in the new log
-Close the log file, and re-open. Parse fails.

The corrupted log file contains duplicate "pressure0" entries in samples

Note that merging dives 475 and 476 does not result in this bug.

### Current behavior:
Log file corrupted.

### Expected behavior:
Not corrupted :)

### Additional information:

### Mentions:

bstoeger, 2020-06-28:
I can reproduce this - will look into it later.

bstoeger, 2020-06-28:
Two remarks:

1) the merging is clearly buggy. The resulting profile looks very odd.

2) The corruption of the log can be prevented by the following patch, however I have no idea if this does the right thing:
```
diff --git a/core/save-xml.c b/core/save-xml.c
index 54469a1cf..c3b9607c8 100644
--- a/core/save-xml.c
+++ b/core/save-xml.c
@@ -268,7 +268,7 @@ static void save_sample(struct membuffer *b, struct sample *sample, struct sampl
                }
 
                /* The new-style format is much simpler: the sensor is always encoded */
-               put_format(b, " pressure%d=", sensor);
+               put_format(b, " pressure%d=", idx);
                put_pressure(b, p, "'", " bar'");
        }
```

@dirkhh? @torvalds?

torvalds, 2020-06-28:
No, that patch can't right. It loses the sensor data entirely.

Let me take a look.

torvalds, 2020-06-28:
I think the problem is that we don't update the sensor numbers when we merge the dives. 

We have clearly taken two dives with two cylinders each, and merged them into one dive with four cylinders. But then we haven't renumbered the sensors from the second dive to match the cylinders from the second dive.

Strange. We shouldn't merge two dives on top of each other at all with just one divecomputer. They should be sequential, not "merge each sample together. 

So there's something really odd going on. I'm not seeing what the bug is, but it's clearly a bug.

torvalds, 2020-06-28:
Oh. We *are* making them sequential. The source file says

  <sample time='0:30 min' depth='3.78 m' pressure0='248.901 bar' pressure1='243.385 bar' />

but the resulting merged file says

  <sample time='0:30 min' depth='3.78 m' pressure0='248.901 bar' pressure0='243.385 bar' />

so we've literally _lost_ the sensor information when we did the merge.  It's not that we've merged the sensor information wrong, it's that we've dropped it entirely somehow.

Hmm. Will have to go off right now, but I'll come back to this.

torvalds, 2020-06-28:
Ok, I think I see the problem.

I'm not sure it ever worked, but commit 7c9f46acd202 ("Core: remove MAX_CYLINDERS restriction") definitely did some odd things. In particular, see merge_cylinders(), where we're supposed to have walked through the cylinder indexes that dive a/b used, and created a mapping[] array of old->new sensor in "mapping_a[]" and "mapping_b{}".

But it has this insane line in it:

                mapping_a[i] = res->weightsystems.nr;
...
                mapping_b[i] = res->weightsystems.nr;

which makes no sense. 

The code *used* to do

-               mapping_a[i] = i;
-               mapping_b[i] = -1;
..
-               mapping_b[i] = j;

to create that mapping opf cylinder sensor indexes for dive a/b respectively. But that logic got corrupted entirely.

bstoeger, 2020-06-28:
> mapping_a[i] = res->weightsystems.nr;

Ouch. :-P

bstoeger, 2020-06-28:
This seems to (at least superficially) fix the problem:
```
diff --git a/core/dive.c b/core/dive.c
index 60b7d1faa..f0c812aa9 100644
--- a/core/dive.c
+++ b/core/dive.c
@@ -2327,7 +2327,7 @@ static void merge_cylinders(struct dive *res, const struct dive *a, const struct
        /* Now copy all the used cylinders from 'a' that are used but have not been matched */
        for (i = 0; i < a->cylinders.nr; i++) {
                if (used_in_a[i]) {
-                       mapping_a[i] = res->weightsystems.nr;
+                       mapping_a[i] = res->cylinders.nr;
                        add_cloned_cylinder(&res->cylinders, *get_cylinder(a, i));
                }
        }
@@ -2335,7 +2335,7 @@ static void merge_cylinders(struct dive *res, const struct dive *a, const struct
        /* Finally, copy all the used cylinders from 'b' that are used but have not been matched */
        for (i = 0; i < b->cylinders.nr; i++) {
                if (used_in_b[i]) {
-                       mapping_b[i] = res->weightsystems.nr;
+                       mapping_b[i] = res->cylinders.nr;
                        add_cloned_cylinder(&res->cylinders, *get_cylinder(b, i));
                }
        }
```

torvalds, 2020-06-28:
That's at least somewhat more sane, but I think the whole logic is broken.

I pushed out a rewrite of this whole logic, but I have _not_ thought about it to a huge degree.

See

    https://github.com/subsurface/subsurface/pull/2888

does that make sense to people?

torvalds, 2020-06-28:
I'm getting some eror emails from that pull request of mine, but nothing showing on the web interface.

I'm taking a break from this now. Maybe Berthold's minimal fix is sufficient, but I think that commit 7c9f46a ("Core: remove MAX_CYLINDERS restriction") was also broken when it removed the logic for "already tried to merge this cylinder", and the code to check whether the cylinder information was filled in (not just "actively used").

mwerle, 2020-07-15:
> the code to check whether the cylinder information was filled in (not just "actively used").

I'm not sure that ever worked for sidemount. 
Once I got sensors for my sidemount cylinders, Subsurface correctly auto-added 2 cylinders, but only marked one as actively used despite the divecomputer log showing gas being depleted from both at various stages throughout a dive.

bstoeger, 2020-10-03:
Should be fixed by #2888. Closing - reopen if problem still exists.

closed 2020-10-03 by bstoeger