subsurface/subsurface #2887: mobile website ios: hamburger menu not working inlanding page

Ospeleo, 2020-06-26:
### Describe the issue:
On iOS 13.5.1, safari, iphone Xsmax, it is impossible to browse the website: the hamburger menu is not working.

We are left with the landing page with the latest version annoucement, can tap to changer langage, but that's all

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

Ospeleo, 2020-06-26:
I just checked on my desktop, and if I reduce the browser window size, the hamburger menu is not working either.

dirkhh, 2020-07-02:
Nah, that was the wrong issue that was closed. The website is still broken and likely will remain so as the underlying infrastructure is broken and I have neither time nor energy to fix it.

joeylobianco-better, 2022-08-11:
> Nah, that was the wrong issue that was closed. The website is still broken and likely will remain so as the underlying infrastructure is broken and I have neither time nor energy to fix it.

Where is this broken infrastructure detailed out, so that others can try to fix it?

dirkhh, 2022-08-11:
The old part of the website (mainly the landing page and a couple of older other pages and the manuals) are in a static export of an ancient broken WordPress site. All the new stuff is in GitHub pages. So what's missing is a port of the landing page to GitHub pages...

mikeller, 2026-10-17:
Hi @Ospeleo,

Good news, this is resolved. The website has since been completely rebuilt (the old landing page was a static export of a legacy WordPress site, which is what dirkhh was referring to as "broken infrastructure" back in 2022). The current site no longer uses a hamburger menu at all, navigation is an always-visible sidebar/icon-pill menu, confirmed by checking the live site and its markup directly today.

Closing this as fixed by the subsequent site rebuild. Please reopen if you're still seeing a non-functional hamburger menu (or any other navigation problem) on a current visit to subsurface-divelog.org.

commit fb420d89 (2020-06-29): profile: fix displaying of profiles with multiple pressure sensors

When removing the MAX_CYLINDERS restriction, the layout of the
pressure readings was changed from a (cylinder,sample) to a
(sample,cylinder) scheme. I.e. previously there were one cylinder
block for each sample, then one sample block for one cylinder.

However, after populating the samples, the array size was reduced
to the actual number of used samples. With the new layout this
breaks indexing. Therefore, restore the old layout.

Fixes #2887
commit 4ec88aa5 (2020-07-02): profile: fix displaying of profiles with multiple pressure sensors

When removing the MAX_CYLINDERS restriction, the layout of the
pressure readings was changed from a (cylinder,sample) to a
(sample,cylinder) scheme. I.e. previously there were one cylinder
block for each sample, then one sample block for one cylinder.

However, after populating the samples, the array size was reduced
to the actual number of used samples. With the new layout this
breaks indexing. Therefore, restore the old layout.

Fixes #2887
closed 2026-10-17 by mikeller