subsurface/subsurface #2890: Subsurface-4.9.4-x86_64.AppImage -- Failed to claim the usb interface (LIBUSB_ERROR_BUSY)

darrell-barabash, 2020-06-29:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Unable to connect to Scubapro G2 via USB when running the AppImage. Connection works fine when running the binary. Problem with the binary is that the font properties are not properly updated (they are saved OK but don't get applied unless the font size is toggled). On the other hand, the fonts work fine in the AppImage but the USB interface does not.

The basic troubleshooting I was able to do:

**OK => Binary Installation Subsurface 4.9.4 with G2 CONNECTED TO USB**
- Starting download from  
- downloading only new dives
- INFO: dc_device_open error value of 0
- Finishing download thread: 29 dives downloaded

From these next tests I looks like the G2 USB connection is actually being detected (state changes when G2 is connected versus disconnected). But, when connected there seems to be "failed to claim the USB interface" error message is coming from libdivecomputer (I think ...)

**NOT OK FOR THIS SCENARIO => AppImage Subsurface 4.9.4 with G2 CONNECTED TO USB**
- Starting download from  
- downloading only new dives
- [0.018306] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in /__w/subsurface/libdivecomputer/src/usbhid.c:607 (dc_usbhid_open)]
- Finishing download thread: "Unable to open  Scubapro (G2)"

**OK FOR THIS SCENARIO => AppImage Subsurface 4.9.4 with G2 DISCONNECTED**
- Starting download from  
- downloading only new dives
- Finishing download thread: "Unable to open  Scubapro (G2)"

**OTHER COMMENTS**
* I have installed a udev rules file and the binary works properly as a normal user.
* I have tried running the AppImage as a normal user. Also tried running as root (on a different machine since it is discouraged to do this) but problem is still the same.
* I have tried running this AppImage on three different computers but all are running MINT 19.3 or MINT 19.2. Same problem.

### Operating system:

Linux MINT 19.3
Linux 5.3.0-61-generic #55~18.04.1-Ubuntu SMP Mon Jun 22 16:40:20 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Scubapro G2 computer
ATTRS{idVendor}=="2e6c",ATTRS{idProduct}=="3201"

### Subsurface version:

This version has USB error.
**Subsurface-4.9.4-x86_64.AppImage**
Subsurface v4.9.4,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (7882ba423cd87569f9c9361fa999917e9ced1083)
built with Qt Version 5.12.4, runtime from Qt Version 5.12.4
built with libgit2 0.26.0

With this version the font properties are not getting set correctly but the USB interface to the Scubapro G2 works OK. 
**subsurface binary v4.9.6 from PPA**
using qt5ct plugin
Subsurface v4.9.6,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (48e46cf7775452740f67ba72d9c5a0960ebf3718)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 1.0.0

### Steps to reproduce:

1) connected G2 to USB port and entered 'lsusb' to make sure device it was detected
2) ./Subsurface-4.9.4-x86_64.AppImage
3) Issue is repeatable and occurs every time. Issue can be repeated on a different computer running MINT 19.3 as
well as a computer running MINT 19.2

### Current behavior:

Starting download from  
downloading only new dives
[0.018306] ERROR: Failed to claim the usb interface (LIBUSB_ERROR_BUSY). [in /__w/subsurface/libdivecomputer/src/usbhid.c:607 (dc_usbhid_open)]

### Expected behavior:

- Starting download from  
- downloading only new dives
- INFO: dc_device_open error value of 0
- Finishing download thread: 29 dives downloaded

### Additional information:

### Mentions:

sparkprime, 2020-08-10:
I got this on chromebook and Ubuntu for a Suunto D5.  Also using the appimage.  Is it a workaround to use another release (not appimage)?  I'm guessing the appimage was built against a buggy libusb or something like that?

darrell-barabash, 2020-08-10:
I'm am wondering if it might be some appimage limitation? I don't think it is permissions because it doesn't work even after creating a udev file. I even tried the "highly discouraged practice" of running as root, just to see if permissions, and that didn't help either. But, USB in an older binary version (as described above) works fine. The error appears to be coming from libdivecomputer but I'm not smart enough to dig deeper and determine root cause.

The only work around I found is to use either my Android phone or an older binary of Subsurface to read the dive computer. Then switch to the Appimage to manage the log.

Thanks for report this @sparkprime ... this says it's not just me having the issue. I was starting to wonder. ;-)

darrell-barabash, 2020-08-10:
Oops ... didn't mean to close this.

sparkprime, 2020-08-11:
I had a separate permissions issue that I "fixed" by changing the permission of `/dev/usb/hiddev1` manually.  Before doing that I got a different error, then I got this BUSY error instead.  So definitely not permissions.

I haven't been able to get it working on android either.  It detects the dive computer but then doesn't download anything when I click download.

vavincavent, 2020-08-30:
I have the same issue with :

- Subsurface-4.9.6-x86_64.AppImage
- Debian GNU/Linux 10 (buster)
- Aladin square

Vavincavent

hughesjs, 2024-05-24:
I'm still having this issue, any ideas?

mikeller, 2024-05-24:
@hughesjs: What version of the AppImage are you using, and is this the same error as reported above?

hughesjs, 2024-05-24:
> @hughesjs: What version of the AppImage are you using, and is this the same error as reported above?

`6.0.5112-1` and yes, but I've swapped to main, built it myself, and now I'm not having this problem

mikeller, 2024-05-25:
> `6.0.5112-1` and yes, but I've swapped to main, built it myself, and now I'm not having this problem

Thanks for the feedback. I'd be keen to hear if the problem still exists when you try with the latest AppImage from https://www.subsurface-divelog.org/latest-release/.

hughesjs, 2024-05-31:
> > `6.0.5112-1` and yes, but I've swapped to main, built it myself, and now I'm not having this problem
> 
> Thanks for the feedback. I'd be keen to hear if the problem still exists when you try with the latest AppImage from https://www.subsurface-divelog.org/latest-release/.

I'll try to remember to give this a try and get back to you

mikeller, 2026-10-18:
Hi @darrell-barabash,

Following up on this one: the fact that a from-source build works fine while the official AppImage specifically fails with LIBUSB_ERROR_BUSY points to this being an AppImage packaging limitation (most likely a bundled libusb version or the AppImage's FUSE-mounted sandboxing interacting badly with direct USB device claims) rather than something we can cleanly fix in Subsurface's own code.

Rather than invest further in the AppImage packaging specifically, we'd recommend switching to the official Flatpak, it's now our actively maintained, recommended Linux package, and doesn't carry this specific limitation.

Closing this as an AppImage-specific packaging constraint rather than a code defect. Please reopen if the Flatpak exhibits the same USB-busy error for your Scubapro G2.

closed 2026-10-18 by mikeller