subsurface/subsurface #2892: What is GF (not Surface GF) shown in the Information Box

emmdarakis, 2020-06-30:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Towards the end of most of my dives an extra line appears in the Information Box called "GF" followed by a percentage that changes as I move the cursor further. I can't think of what this might be and nothing relevant is mentioned in the Manual hence I suspect this is a bug. Notice that I am not talking about "Surface GF" which is shown as it should. 

### Operating system:
Windows 10 Pro, build 18362.900 in PC

### Subsurface version:
4.9.6 official release. I am pretty sure I have seen this for several previous releases (including test) but I don't know since when. 

### Steps to reproduce:
As the cursor moves towards the end of most of my dives, an extra line appears in the Information Box called "GF" followed by a percentage. Notice that I am not talking about "Surface GF" which is shown as it should be. 
I am attaching two screen shots. One with the information box as I think it should be ( @50:30 of my dive) and one with the extra "GF" line which I highlight in red (@ 52:57)

[image]
[image]

It might be something useful that I am not aware of (in which case I apologize), but in anyway nothing relevant is mentioned in the manual.

atdotde, 2020-06-30:
This is your current gradient factor (current as in: compared to ambient pressure and not surface pressure).

closed 2020-06-30 by atdotde

emmdarakis, 2020-06-30:
Ok, makes sense. Learning something new every day.
Thanks and sorry for this...