subsurface/subsurface #2901: Data error when Planning and saving dive with surface time

Ospeleo, 2020-07-23:
Bug
- Context: using the planner to plan a multi-sump dive for cave diving
It is useful to plan a multi sump dive as one single dive for deco/gaz management

- Error: If you manually enter dive points with return to surface (depth=0, with/without surface time), you save it, then the dive is correctly displayed.
If you re-edit it, then Subsurface misreads the saved file and all the zero-depth points are merged, giving a wrong dive profile.

Subsurface 4.9.6 macos 10.12.6

screen captures:
Planning the dive (entering manual dive points)
[image]

The dive as it appears in the main window after being saved. It looks good
[image]

Re-editing the same dive: all zero-depth points have been merged
[image]

atdotde, 2020-08-17:
The problem is not zero depth but zero length segments (instantaneous transitions to new depth). Our computational model doesn't like you to be at two depths at once and thus deletes those waypoints. You could argue that the failure should be more gracefully, for example by adding a few seconds for the transition and I will submit a patch to that effect.

Ospeleo, 2020-08-17:
thanks. it makes sense. 
instead of a few seconds, you could just add a millisecond so that it's not the same time.

commit 62d87e9d (2020-08-24): Planner: handle zero length segments when replanning

When setting up a dive for replanning, we ignored zero length segments as those
tend to be generated by gas changes. But it is possible to enter those in the
planner and the replanning should not ignore those. So be
more clever about gas changes. Let's add 10 seconds so we are not at two depths
at the same time and help since add_stop also does not like zero length
segments (it thinks we are trying to replace a waypoint).

Fixes #2901
closed 2020-08-24 by dirkhh