subsurface/subsurface #2903: Cylinder without gas change not shown after reload

gadton, 2020-07-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
If you add a second cylinder to a dive but you do not have a gas change event, the cylinder is stored but at reloading the file not shown any more. Adding a second cylinder the second time results in three <cylinder ...> entries in the file but still onyl one after reloading.

### Operating system:
Win 10
German

### Subsurface version:
4.9.5

### Steps to reproduce:
1) import dive from computer (without gas integration)
2) add second cylinder to dive X (e.g. just for information because it was a SM dive)
3) save file
4) close Subsurface
5) open Subsurface
6) select dive X

### Current behavior:
* table of cylinders shows only ONE cylinder
* file contains two cylinders

### Expected behavior:
second cylinder is shown in table

mturkia, 2020-07-31:
This is by design as some divecomputers report multiple un-used cylinders. If you wish to see un-used cylinders, there is configuration option under Equipment. (File -> preferences -> Equipment)

gadton, 2020-07-31:
Thank you for the explanation.

closed 2020-07-31 by gadton