subsurface/subsurface #2907: Bluetooth Import Fehler

123kub, 2020-08-09:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Fehler beim Import der Tauchgangsdaten.

Wenn ich die TG-Daten "erneutes Herunterladen erzwinge" , dann werden alle korrekt geladen.

### Operating system:
 Win 10
 Shearwater Perdix AI

### Subsurface version:
 4.9.6
 YES
YES

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-08-09:
Ich sehe gelegentlich Timeouts und Errors beim BLE Herunterladen. Und dann klappt es mit einem neuen Versuch.
BLE Verbindungen sind immer etwas zerbrechlich. Und waehrend der Tauchcomputer Anbieter natuerlich beide Seiten der Kommunikation kontrolliert, versuchen wir, mit vielen verschiedenen Tauchcomputern zu arbeiten. Da kommt sowas gelegentlich vor.
Solange es durch erneutes Herunterladen behebbar ist, sehe ich das nicht als katastrophal an.

123kub, 2020-08-09:
Hallo.
Naja. Ich sehe das schon etwas anders. Ich habe bisher "Nur" ca 150 Tauchgänge zu übertragen. Das dauert jedoch ca 1 Stunde !
Wenn ich dann JEDESMAL erneut ALLE Tauchgänge übertragen muss, dann wird das irgendwann schnell mal 2 Stunden dauern ... Sorry, aber das finde ich nicht ok, da es ja mal funktioniert hatte ( ich weiß leider die Versionsnummer nicht mehr ). Bis 4.9.3 habe ich schon "downgrade" gemacht.... aber keinen Erfaolg. Den TC kann ich nicht auf eine andere Version flashen ( könnte nur der Hersteller in Werk ).

dirkhh, 2020-08-09:
Nein, das sollte nicht so sein.
Normalerweise uebertraegt man ja nur die Tauchgaenge eines Tages (oder vielleicht von ein paar Tagen).
Und das Herunterladen sollte in der Regel funktionieren. Und falls es nicht funktioniert, dann kann man einfach die letzten paar Tauchgaenge loeschen und das ganze "normal" nochmal versuchen.

Und zur Aussage "das finde ich nicht ok"... ich werde gerne den Kaufpreis zurueckerstatten.
Oh, das ganze war umsonst? Naja, dann vielleicht die Anspruchshaltung und den Ton etwas ueberdenken?

Deutsche. Was soll man da sonst sagen.

123kub, 2020-08-09:
Das ist aber so ...
Schließlcih habe ICH , das ganze gerade durchgemacht !!
hat gute 2,5 Stunden gedauert !!!
Und nicht nur die letzten 2 TG !!

Ich ob ich es OK finde oder nicht das ist allein mein Empfinden
Jeder hat das Recht auf SEINE EIGENE Meinung !!!

Und die schnippische Antwort mit dem Kaufpreis .... Das können Sie sich echt sparen !!!
Und ebenfalls können Sie IHREN Ton einmal vielleicht überdenken.

Ich mache auch mehrer Sachen ( Jungenprojekte ) KOSTENLOS und Freiwillig ( Wie Sie  ) !!
und habe noch nie gesagt ... wenn es Dir nicht gefällt,dann zahl mir die Unterbringsgebühr
oder die Kosten für das Material zurück.

Sorry, aber das ist eine ABSOLUT unqualifizierte Aussage / Antwort !!!

Sie hätten noch schreiben sollen, das es noch andere Anbieter gibt, und es mir freisteht, Ihre Software zu nutzen,
oder zu anderen zu wechseln.
Dann wäre es das Maximalste gewesen !!

Und was soll das mit "Deustche. Was soll man dazu sagen" ???

Sind Sie etwas ein Rassist ???!!
Ich wäre damit sehr vorsichtig mit solchen Aussagen !!
Das kann schnell in einer Strafanzeigen enden.
( Das nur als Hinweis ) !!!

dirkhh, 2020-08-09:
Good bye. When you think you can threaten me we are done here.

closed 2020-08-09 by dirkhh