subsurface/subsurface #2908: Installation of subsurface on Fedora 32 not working.

thommmmmy, 2020-08-09:
### Describe the issue:

Installation of subsurface on Fedora 32 not working. The RPM-Package depends on out of date libraries. 
- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I try to run subsurface on Fedora 32 (x86-64)

Since subsurface isn't part of Fedora itself any more, it isn't working seamless. 

Read https://bugzilla.redhat.com/show_bug.cgi?id=1742889 
This worked well for Fedora 30. Right after the release of it was working for Fedora 32 as too. But then is brakes despises. 

The current RMP-Package ( https://download.opensuse.org/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm ) is out of date. It depends on old libraries. 

If I downgrade the affected libraries, the functionality of many qt- and KDE bases apps are broken. (E.g. dolphin, VLC, HandBrake,  kdenlive and more stops working!)

**IMPORTANT:**
I was advised to use the Flatpak-Package of subsurface more then once in the past. But there is one big issue:  Flatpak is in this case no option, because I need to be able to import data form the  commuter (over USB).

dnf trying to install subsurface on Fedora 32: 

> **root@localhost ~ $**
> 
> **root@localhost ~ $ wget https://download.opensuse.org/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm**
> --2020-08-09 11:50:29--  https://download.opensuse.org/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm
> Resolving download.opensuse.org (download.opensuse.org)... 2001:67c:2178:8::13, 195.135.221.134
> Connecting to download.opensuse.org (download.opensuse.org)|2001:67c:2178:8::13|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://ftp.gwdg.de/pub/opensuse/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm [following]
> --2020-08-09 11:50:29--  https://ftp.gwdg.de/pub/opensuse/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm
> Resolving ftp.gwdg.de (ftp.gwdg.de)... 2001:638:60f:110::1:2, 134.76.12.6
> Connecting to ftp.gwdg.de (ftp.gwdg.de)|2001:638:60f:110::1:2|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 7974100 (7.6M) [application/x-redhat-package-manager]
> Saving to: ‘subsurface-4.9.6-2.1.x86_64.rpm’
> 
> subsurface-4.9.6-2.1.x86_64.rpm                   100%[==========================================================================================================>]   7.60M  5.75MB/s    in 1.3s    
> 
> 2020-08-09 11:50:30 (5.75 MB/s) - ‘subsurface-4.9.6-2.1.x86_64.rpm’ saved [7974100/7974100]
> 
> **root@localhost ~ $ dnf install ~/subsurface-4.9.6-2.1.x86_64.rpm** 
> Last metadata expiration check: 2:51:37 ago on Sun 09 Aug 2020 08:59:43.
> Dependencies resolved.
> =====================================================================================================================================================================================================
>  Package                                                   Architecture                       Version                                               Repository                                  Size
> =====================================================================================================================================================================================================
> Installing:
>  subsurface                                                x86_64                             4.9.6-2.1                                             @commandline                               7.6 M
> Downgrading:
>  HandBrake                                                 x86_64                             1.3.1-1.fc32                                          rpmfusion-free                             438 k
>  HandBrake-gui                                             x86_64                             1.3.1-1.fc32                                          rpmfusion-free                             3.5 M
>  digikam                                                   x86_64                             7.0.0-0.3.beta2.fc32                                  fedora                                     104 M
>  digikam-doc                                               noarch                             7.0.0-0.3.beta2.fc32                                  fedora                                     8.0 k
>  digikam-libs                                              x86_64                             7.0.0-0.3.beta2.fc32                                  fedora                                     9.9 M
>  ffmpeg                                                    x86_64                             4.2.2-5.fc32                                          rpmfusion-free                             1.4 M
>  ffmpeg-libs                                               x86_64                             4.2.2-5.fc32                                          rpmfusion-free                             7.1 M
>  gpgme                                                     x86_64                             1.13.1-6.fc32                                         fedora                                     205 k
>  gpgmepp                                                   x86_64                             1.13.1-6.fc32                                         fedora                                     133 k
>  kdenlive                                                  x86_64                             19.12.3-1.fc32                                        rpmfusion-free                              13 M
>  kf5-akonadi-server                                        x86_64                             19.12.2-1.fc32                                        fedora                                     2.5 M
>  kf5-akonadi-server-mysql                                  x86_64                             19.12.2-1.fc32                                        fedora                                      11 k
>  kf5-frameworkintegration                                  x86_64                             5.68.0-1.fc32                                         fedora                                     1.6 M
>  kf5-frameworkintegration-libs                             x86_64                             5.68.0-1.fc32                                         fedora                                      29 k
>  kf5-kdeclarative                                          x86_64                             5.68.0-1.fc32                                         fedora                                     293 k
>  kf5-ktexteditor                                           x86_64                             5.68.0-1.fc32                                         fedora                                     2.5 M
>  kf5-kwayland                                              x86_64                             5.68.0-1.fc32                                         fedora                                     516 k
[... 99 lines cut ...]
> Verifying...                          ################################# [100%]
> Preparing...                          ################################# [100%]
> Updating / installing...
>    1:subsurface-4.9.6-2.1             ################################# [100%]
> /var/tmp/rpm-tmp.X3TfGf: line 1: fg: no job control
> **root@localhost ~ $ exit**
> **thom@localhost ~ $ subsurface**
> subsurface: error while loading shared libraries: libgit2.so.0.99: cannot open shared object file: No such file or directory
> **thom@localhost ~ $** 

Subsurface won't load. Not working.

### Current behavior:
- Installation with dnf isn't working as it should. Downgrading of 77 core-packages is a now go!
- Flatpak package doesn't offer access to the dive computer. 
- After forcing the installation, subsurface won't start.

### Expected behavior:
Two options:
1. Fix despises in the package at: https://download.opensuse.org/repositories/home:/Subsurface-Divelog/Fedora_32/x86_64/subsurface-4.9.6-2.1.x86_64.rpm
Or
2. Make dive computer data import over USB working in the Flatpak package. 

### Mentions:
I really like subsurface, if I'm able to run it.

dirkhh, 2020-08-12:
What an exhaustingly extensive bug report. Thanks for providing ooooooodles of details, @thommmmmy 
I'm talking to the open build service folks to understand why their Fedora 32 builds are linking against old libraries. Hopefully this can be fixed fairly easily.

dirkhh, 2020-08-18:
Sorry this took so long. I finally figured out how to fix this. In hindsight the fix is of course simple and obvious, but... yeah.
Anyway, the updated RPMs should have propagated to the mirrors, version 3.9.6-4.1 should be available for Fedora 32 and I verified that it installs cleanly.

Thanks for your report!

closed 2020-08-18 by dirkhh