subsurface/subsurface #2909: Scubapro Aladin Sport Matrix version 2.1 - no longer possible to connect to Subsurface

patriziodazzi, 2020-08-12:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After upgrading the firmware of Scubapro Aladin Sport Matrix to version 2.1, it is no longer possible to connect to Subsurface to download the dives.

### Operating system:

macOS Catalina 10.15.6 (19G73)

Scubapro Aladin Sport Matrix

### Subsurface version:

4.9.6

### Steps to reproduce:

Connect Aladin Sport Matrix firmare v2.1 to Subsurface and try to download new dives

### Current behavior:

An error raises saying: Error in importing details of the dive

### Expected behavior:

To download the data of the dive

### Additional information:

### Mentions:

bstoeger, 2020-08-21:
Stupid question - how do you update the firmware of the Aladin Sport Matrix?

info-kubu-at, 2020-11-12:
see https://ww2.scubapro.com/logtrak/updating_aladin_matrix_software.pdf

info-kubu-at, 2020-11-12:
but i downloaded already many dives with the new software-version 2.1 on a android device

Taidow3, 2021-06-27:
Any news on this? Planning to buy the Aladin Sport Matrix, if it works with Subsurface.

bstoeger, 2021-06-27:
Sorry, I totally forgot about that.

Sadly, I can't download the V2.1 firmware. The link on this page: https://www.scubapro.eu/software-firmware is dead for me.

Taidow3, 2021-06-27:
Their link is wrong:

https://ww2.scubapro.com/logtrak/AladinMatrix_V21%20REL.swu (404)

the correct one is:

https://ww2.scubapro.com/logtrak/AladinMatrix_V21.swu (%20REL removed)

bstoeger, 2021-06-27:
Sorry, I can't get my computer updated. After numerous attempts, the farthest I've reached is 86% with my phone and 2% (sic!) with my tablet. :(

Taidow3, 2021-06-27:
Thank you so much for trying. I might just get a different computer. With the wrong download link and update difficulties it doesn’t seem the Aladin Sport Matrix is not the most reliable/polished computer ;)

bstoeger, 2021-06-27:
To be fair, it served me (a very casual holiday diver) well.

I finally managed to update the firmware. I guess the trick was to tap on the phone screen every few seconds for half an hour to avoid Android putting the app into sleep mode.

After the firmware update I had to unpair/pair on Android to get LogTrak to connect - so perhaps (hopefully) that is the problem of the reporter. I'll investigate  tomorrow (I don't have my laptop from which I usually download dives with me).

bstoeger, 2021-06-28:
Indeed, after the Firmware update I couldn't immediately connect.

What I had to do was to pair the device using `bluetoothctl`. I did not find a way to do this from KDE/Plasma.

Then it did work, however. I wonder if this bug is valid?

One strange thing:
On one laptop I got a sensible message (needs PIN, therefore pair with the OS).
On another laptop I got an unhelpful message (unknown error).
Both run Kubuntu 21.4. So it seems to depend on the Bluetooth driver?

@dirkhh: should we reverse-engineer the pairing code of `bluetoothctl` and reimplement it, or should we just assume that the average Linux user can do this for themselves?

bstoeger, 2021-06-28:
Update: I ultimately also managed to pair the computer with my other laptop (for some reason that was a bit more painful). Then the connection seems to work just fine.

I feel that this bug report is obsolete. Close?

dirkhh, 2021-06-28:
I think we shouldn't try to reimplements that part of pairing (we do support Qt-level pairing in the BT connection screen.
There are two many variations across our different OSs

closed 2021-06-28 by dirkhh

bstoeger, 2021-06-28:
> we do support Qt-level pairing in the BT connection screen.

which sadly doesn't support PIN pairing on Linux based operating systems. The correct place to fix this would probably be Qt. Good luck with that. :(

> There are two many variations across our different OSs

Right - I reckon that including bluez in the build system alone would be pure hell.

Taidow3, 2021-06-28:
Thanks for your effort. Will probably buy the computer and try the connection with macOS.