subsurface/subsurface #2911: Please add bluetooth support for Shearwater Peregrine (iOS)

eliempje, 2020-08-14:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:
New Dive Computer by Shearwater has been released recently: Shearwater Peregrine. I would like to use this with Subsurface (iOS) app.

### Operating system:
iOS mobile app

### Steps to reproduce:
Connecting the Shearwater Peragrine is currently not possible.

### Current behavior:

Currently unable to select this model from Shearwater.

### Expected behavior:

Make Shearwater Peragrine an option to select as bluetooth device to connect to.

### Additional information:
I'm happy to support this change (even by coding) and test/debug this functionality.

dirkhh, 2020-08-14:
This is actually in the works. I have a Peregrine and am in contact with Shearwater, but for unexplained reasons the Peregrine currently doesn't want to talk to our BLE code. Not sure how long it will take to figure this out, TBH.

dirkhh, 2020-08-18:
Quick update. There is an odd issue with the Peregrine BLE implementation. Shearwater's own app doesn't stumble over this, but since we are using a standard toolkit to talk to BLE devices, we can't connect to the Peregrine right now.
I'm talking to Shearwater and assume that we'll come up with a solution for this, but right now this is on pause.

eliempje, 2020-08-19:
Thank you. Indeed I had the same experience. Let me know if I can assist/help/test.

dirkhh, 2020-09-13:
this is now available on both iOS and Android and will be in the next Subsurface version as well

closed 2020-09-13 by dirkhh

eliempje, 2020-09-13:
Great!