subsurface/subsurface #2918: Independent Cylinders data missing - Perdix AI

ryanm101, 2020-08-19:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I'm not sure but it looks like there are a couple of issues:
1. SAC is not being calculated on all parts of the graph
2. Cylinder data on graph is not displayed correctly based on times.
3. Cylinder data is missing from the SSRF output that i can see the data is present in the Shearwater Cloud Desktop export (Note I imported data directly from Perdix AI in both cases, I did not export and import between applications)

### Operating system:
Mac/Fedora32

### Subsurface version:
4.9.6

### Steps to reproduce:
Import Dive from Perdix AI where two cylinders have been used independently during the dive (sidemount)

### Current behavior:;
See Image,
MAC:
[image]
Notice the 2nd cylinder starts at 19min after my first. yet i switched at approx 7min also the grey line between 08:30 and 13:00 (I didn't become a fish for 5min, though the high sac at 7min was me attempting to breath water, unsuccessfully i might add)
linux:
[image]

### Expected behavior:
As I am only breathing off 1 cylinder at a time i should see either:
1. -_-_____- where the profiles are inverted as if running concurrently (i.e. motion on one line while the other is flat)
2. (ideally)The software should be able to determine which cylinder is being breathed off and should 'auto' switch cylinders on the profile with expected mistakes for buoyancy when using the opposite cylinder.
3. Cylinder pressure data should be present in SSRF similar to shearwater XML (is this maybe 'deduplication'?)
4. SAC should appear over all parts of the line 

### Additional information:
I've included the shearwater desktop client output for comparison, as you can see looking at the XML file tank pressure is recorded every 2sec where as this is missing in a number of entries for Subsurface.  

These were 'clean' imports to new logbooks for the purpose of this post. 
2x 10L 232 cylinders
[Archive.zip](https://github.com/subsurface/subsurface/files/5095249/Archive.zip)

mwerle, 2020-08-24:
(Subsurface 4.9.6-41-g2a09270f270a  Linux self-compiled; same issue with official Subsurface 4.9.0, Debian GNU/Linux)

Looking at your data, it's not missing, it's just not being graphed correctly.

Going through my own dives, I've also found a couple with this issue. Other dives are graphed correctly.  I can't see (from a data perspective) why some dives are graphed correctly and others are not.

Dive with data not graphed correctly (grey zones are where I was breathing from the other tank):
[image]

Dive with data graphed correctly:
[image]

[Dives.zip](https://github.com/subsurface/subsurface/files/5120390/Dives.zip)

ryanm101, 2020-09-10:
interestingly the latest build from master (4.9.6-72-gbeedea7ccfc) does seem to have fixed the graphing issue, may be coincidence as @mwerle had some that were affected and some that weren't

ryanm101, 2020-09-28:
Fixed in 4.9.7 for me.

closed 2020-09-28 by ryanm101